ant-thomas/zsgx1hacks #103: when wifi connection is lost camera remains disconnected digoo w01f

gumslone, 2019-02-22:
Hi, 

i've noticed that my cameras don't try to reconnect to wifi when the wifi connection is lost, or when i temporarily reboot my wifi router the cameras remain offline, so i have to disconnect and reconnect the cameras from power source to put them online again.

The possible solution would be for camera to ping the router ip, if the ip is not pingable, to reconnect to wifi or to reboot the camera.

The question is: does the hack already include such functionality?

halninekilo, 2019-02-23:
Have multiple GD-SC01's and similar issues - cameras reconnect to wifi about 50 times a day for some reason and just stop after working for 2-3 days. 
A simple reboot only helps in one of ten cases, more likely it will just forget it's wifi credentials alltogether and blabber on about me using my "phone to connect to wifi"...

Starting to suspect that this is by design to force users to at least re-authenticate with the app every few days if the cam itself can't phone back home (blocked by router).

Tried the readonly hack but no dice and I doub't that it would solve the issue...

gumslone, 2019-02-23:
there is no issue if the camera is connected via lan wire,
only if the camera is connected via wifi and for example if i reboot my router, or temporarily disconnect the wifi network, the camera doesn't appear back online. So i have to disconnect the camera from the power source.

ulrichr88, 2022-08-24:
Hy!

I have the same problem with my digoo Cameras.
@gumslone, , did you find a solution?

THX!