ant-thomas/zsgx1hacks #104: Any this device get working?

tonbor, 2019-03-08:
I bought https://m.banggood.com/nl/WIFI-1080P-ONVIF-IP-Camera-P2P-Wireless-IR-Cut-Security-Camera-Night-Vision-Outdoor-p-1408071.html
Thinking it was onvif, it is not, so try to access it this way.

I got the OR sd loaded. Can telnet, can read it, scan ports, but no Access to onvif,  rtps stream. Website see only ptz no video....
-sh: /bin/hostname: not found
# ls /tmp | grep -F 3
3.4.0.1026
SDK10823
# any hints?

gd-99, 2019-03-08:
@tonbor you are lucky, I have just got this same banggood camera working thanks to ant-thomas' zsgx1hacks working. To be honest it worked fine using the bundled cloud service, but I am not very comfortable allowing a hosted service access to my private life. Using zsgxhacks, uncouples my camera from the cloud - thanks ant-thomas!

The magic info I think you need for this device is the rtps port number: 8001!

Note: I have only got this working using a wired connection or with the camera in AP mode. I have not yet found how to connect the camera to my router. So...

Follow the instructions for the read only Newer firmware (readonlyhack-v0.1.zip) which from reading your post I think you have successfully done. Instructions suggest an Android app, but I used IP Cam Viewer Lite [ https://play.google.com/store/apps/details?id=com.rcreations.ipcamviewer&hl=nl ](url)
Open the app; then under Manage Camera add  (+) a Generic URL camera. 
*Change the Name and Make to suit. 
*As a first try for the Model, select Generic RTSP over TCP.
*URL: rtsp://<your_camera_IP_Address>:8001
*User: admin
*Paswd: <leave empty>

Press the Test button should display the image from your camera. Press Save to save the set-up. Use the Android back (<) button to get back to the camera feed screen.

** When in AP mode the camera IP address is 192.168.200.1. In wired mode it is whatever your DHCP (router) supplies.

I have not yet - working on it - got the PTZ controls to work with the app. I need to use an Internet browser to position the camera.

On the PC I have also been able to view the camera image using VLC. Select the streaming option then enter the "rtsp" string as above. I have successfully incorporated this camera into the Windows only iSpy [https://www.ispyconnect.com/](url) application. Here I have also managed to get PTZ working.

Hope this helps.

tonbor, 2019-03-08:
Thanks, I thought  I was stuck because the onvif device manager could not find the camera, not at port 8080 neither at 8001. The ip Address is in the range of my network and more or less static because of binding the mac tot an Address in my router. I will have a try with the tools you mentioned. If I succeed in getting a wifi connection I let you know.

tonbor, 2019-03-08:
Vlc with rtsp is working YES YES, strange that onvif is not recognized……..

tonbor, 2019-03-09:
Vlc with rtsp is working YES YES, strange that onvif is not recognized…….. and I see this hack has no onvif, the maker is working on it he said in december last year, any progress?

gd-99, 2019-03-09:
UPDATE: managed to get ptz working and integrated with the android app - [IP Cam Viewer Lite](https://play.google.com/store/apps/details?id=com.rcreations.ipcamviewer&hl=nl).

By modifying the webui ptz control app, it is possible to emulate the eneo - PXD-2018PTZ1080 camera ptz commands. This will display the rtsp stream overlayed with the ptz controls. The app allows swiping to move the camera, or by pressing arrow keys.

The process I followed:
1. on the SD card under hack/www/cgi-bin create a folder called com
The path should now look like /hack/www/cgi-bin/com
2. in /hack/www/cgi-bin/com create a file called ptz.cgi
This is my version - remove the .txt from the file name :)
[ptz.cgi.txt](https://github.com/ant-thomas/zsgx1hacks/files/2948494/ptz.cgi.txt)
3. replace the SD card in the camera and reboot it.

The camera's ptz can still be controlled from a browser page. Moving on to the app.

In the IP Cam Viewer Lite app:
1. go to manage a camera and add a new camera using the Generic URL option
2. Name: call it whatever you want.
3. Make: tap the find button (right hand side of the input field) and enter eneo, then tap the find button. 
4. select the eneo PXD-2018PTZ1080 camera /model
5. Host IP: enter the camera's IP address
6. Http(s) Port: 8080
7. Port Overides: RTSP 8001
8. User: admin (password empty)
Tap the Test button to see that the settings can retrieve the camera image. Tap Save to save the camera setting.
Get back to the camera view. Selecting the new "eneo" camera should display the cameras image. If not displayed tap the screen to bring up the camera controls. Tap on the PTZ /Preset to overlay the PTZ arrows to control pan and tilt. Or, simply swipe left /right /up /down to pan left /right /up /down.

I am making some progress at getting the camera to connect to my router (wifi client mode), but can only do this if I boot the camera with an Ethernet cable connected. Once connected to the router I can remove the eth0 cable and the camera keeps the wifi connection. However a re-boot will drop the wifi connection and go back into AP mode.

techspy2000, 2019-03-09:
Awesome, thanks!

gd-99, 2019-03-11:
@techspy2000 following the information in [update 13](https://github.com/ant-thomas/zsgx1hacks#2018-02-13---update-13-ant-thomas), using VLC I get audio and video. the URL you need is: `rtsp://<cam_ip_address>:8001/0/av0 `
I can get the "/0/av0 into the android app, but when I tap the speaker icon on the right hand side of the app viewing screen, the app tells me to upgrade to the pro version :(

Android VLC app also works with the suffix /0/av0, but without the PTZ - bit clunky but guess you can always use the browser to PTZ, then go back to VLC to monitor.

Still working at it :)

techspy2000, 2019-03-11:
Thanks for the info. I am using the pro version of IPCam Viewer so I will give it a try and let you know.

paus56, 2019-03-12:
Для полноценной работы ONVIF в камере на GK7102 в файле hwcfg.ini должна быть обязательно строка:
support_onvif = 1
если её нет, то добавьте в файл.

Если после перезагрузки камера входит в режим AP, в файле hwcfg.ini 
найдите и удалите строку:
support_ap_mode = 1 (или 2)

много полезного по этим камерам там-> http://4pda.ru/forum/index.php?showtopic=928641

Google translate:
A string must be present in the hwcfg.ini file on the GK7102 for ONVIF to work properly in the camera:
support_onvif = 1
if it's not there, add it to the file.

If the camera enters AP mode after rebooting, the file hwcfg.ini 
find and delete the line:
support_ap_mode = 1 (or 2)

a lot of good stuff on those cameras there--> http://4pda.ru/forum/index.php?showtopic=928641

gd-99, 2019-03-12:
@pause56 great find, thanks for sharing. I amended my hwcfg.ini file, and happy to report my camera is now auto detected as an onvif device - so much easier to set-up. Tested with tinycam I have the full PTZ control.  I haven't yet tested out what happens when I remove the ap_mode line.

I am now all over [](http://4pda.ru/forum/index.php?showtopic=928641) like a child in a sweet shop. Only wish I understood Russian - Google translate is my friend?

gd-99, 2019-03-13:
Following up and reading through (all 18 pages) of the discussion pointed to by pause56 I have tested out some of the extra functionality for this camera.

With support_onvif = 1 in /homehwcfg.ini, adding:
   `passwd = passw0rd`  - adds the password "passw0rd" to the rtsp stream. When tested both tinycam and vlc needed the password to display the stream.
   `show_osd_time = 1`  - adds an on screen date and time to the rtsp stream.

Using the osd option, for the camera to get the correct date off the Internet, I had to first modify the debug_cmd.sh, found in the root directory of the sdcard, ntpd entry. I copied the line out of the debug_cmd.sh file found in [DBaluxa's fork.](https://github.com/DBaluxa/zsgx1hacks).
Change: `(sleep 20 && /home/busybox/ntpd -q -p 0.uk.pool.ntp.org ) &`
To: `(while true; do sleep 20 && /home/busybox/ntpd -q -p 0.uk.pool.ntp.org; done ) &`

Then to get the correct time zone - I am in the UK - go to /home and edit start.sh:
Find and change: `echo "CST-8" > /etc/TZ`
To: echo `"GMT-0" > /etc/TZ`

Once the camera has rebooted, after a short delay typing `date` at the command prompt should correct to the current date and time. 
*** Caution when editing the start.sh. An error here and it may be possible to stop the camera starting up correctly. I recommend making a copy of this file before editing.

From the discussion page; I came across a way to get a JPEG 320x176 pixel snapshot image of what the camera has in frame through a web browser - worked for me:
`http://<camera_IP>:554/snapshot`

I suspect results will depend on the camera's firmware level. My camera reports it is:
Model - CHW-PTZ
Firmware - 3.4.0.1026
SDK - 10823

Finally someone in the forum provided a list of all the options for hwcfg.ini. I don't think all options will work for all camera's but have included it here as a txt file for reference.
[hwcfg.txt](https://github.com/ant-thomas/zsgx1hacks/files/2962644/hwcfg.txt)

gd-99, 2019-03-14:
Finally got wifi to connect to a hidden SSID without going through the bundled app /cloud service.

Following paus56 pointer I removed `support_ap_mode = 1` from the hwcfg.ini file, but I then had to set-up several other files.

In /home edit the /home/wpa_supplicant.conf file with the wifi settings. A sample file would look like this:
**wpa_supplicant.conf**
```
ctrl_interface=/var/run/wpa_supplicant
update_config=1
network={
ssid="<your_ssid>"
key_mgmt=WPA-EAP WPA-PSK IEEE8021X NONE
pairwise=TKIP CCMP
scan_ssid=1
group=CCMP TKIP WEP104 WEP40
psk="<your_ssid_password>"
}
```
From [this discussion](https://translate.googleusercontent.com/translate_c?depth=1&rurl=translate.google.com&sl=auto&sp=nmt4&tl=en&u=http://4pda.ru/forum/index.php%3Fshowtopic%3D928641%26st%3D180&xid=17259,15700002,15700019,15700186,15700190,15700248,15700253&usg=ALkJrhjcvA3krTBlY1vxqMQ_0hubNJgj-w), create a cls.conf file in the route directory of the sd-card. It should look like this:
**cls.conf**
```
[cls_server]
ip = 192.168.1.xx
gw = 192.168.1.1
netmask = 255.255.255.0
dns = 208.67.222.222
ssid = <your_ssid>
passwd = <your_ssid_password>
```
Finally for some reason¸ on my camera, although I got confirmation the wifi had connected, the camera didn't assign an IP address to wlan0 - my router showed it had associated correctly.

To correct this I edited start.sh by adding the following to the end of the start.sh file
```
if [ -f /media/hack/set-wifi-ip.sh ]; then
    /media/hack/set-wifi-ip.sh
fi
```
Then created the set-wifi-ip.sh file in the sd-card/hack directory. The file contains:
**set-wifi-ip.sh**
```
#!/bin/sh
# Set wifi IP address
# Run from /home/start.sh
/sbin/ifconfig wlan0 192.168.1.xx
```
This way if the set-wifi-ip file mucks up the startup, simply delete set-wifi-ip.sh to go back to normal.

It is a real hack, as I would like to check the wifi had associated before assigning a fixed IP, but my shell scripting is not up to it. So long as the wired connection isn't present, it should work. My camera doesn't get upset if the wired connection is connected and the wifi gets the same fixed IP address - it just favours the wired route.

To quiet the camera down a bit put; voice_volume = 5 in hwcfg.ini.

tonbor, 2019-03-15:
Thanks for the great work!

tonbor, 2019-03-16:
How do you edit, vi is not working the normal way.

gd-99, 2019-03-16:
Not sure I can help. For me; once I have successfully logged in via telnet, I simply use vi </path/file name> in the normal way.

tonbor, 2019-03-21:
Well gd-99 you told us that you got the same IP camera from Banggood as I bought. Well it seems that my home directory is now write protected (putty telnet port 23). How did you manage to write to it? This is my firmware version:
IP CAMERA login: root
Password:
-sh: /bin/hostname: not found
# ls /tmp | grep -F 3
3.4.0.1026
 SDK10823

gd-99, 2019-03-21:
@tonbor if I read your post correctly:-

> P CAMERA login: root
> Password:
> -sh: /bin/hostname: not found
Indicates you have successfully logged into the camera (using telnet /ssh), however by default, on my camera, this lands in root's home directory aka /root. This can be confirmed by typing `pwd` at the command prompt. The shell should respond with /root. On my camera this (/root) directory is read-only. The home directory refereed to in the "hack" is an actual directory in the file system tree called /home. I have to do a `cd /home` to get into the "home" directory. Type `pwd` to confirm this. This "home" directory is writeable on my camera.

Hope this helps.

By the way the error above "-sh: /bin/hostname: not found " can be corrected by editing the profile file found in the sdcard /hack directory.
Find the the line (around line 31 in my editor) and:
Change:
`export HOSTNAME=`/bin/hostname``
To:
`export HOSTNAME=`hostname``
... then reboot /restart the camera.

tonbor, 2019-03-21:
Well thanks  I was already working in the home directory. Hellas I cannot change files, neither add a file in this directory. Saving a changed file gives a error 10L, 0C. Creating a file with VI is not possible. Perhaps I got bad luck with this camera. It cannot connect wireless, only wired. Saving the wireless when this is configured while I am with that terrible app wired connected, is not working…… Perhaps I have a Monday morning Banggood machine. 

 

Is it possible to load another firmware without bricking the machine?

 

Regards,

 

Rob

tonbor, 2019-03-21:
Well thanks  I was already working in the home directory. Hellas I cannot change files, neither add a file in this directory. Saving a changed file gives a error 10L, 0C. Creating a file with VI is not possible. Perhaps I got bad luck with this camera. It cannot connect wireless, only wired. Saving the wireless when this is configured while I am with that terrible app wired connected, is not working…… Perhaps I have a Monday morning Banggood machine. 

Is it possible to load another firmware without bricking the machine?

Regards,

tonbor

tonbor, 2019-03-21:
IP CAMERA login: pi
Password:
Login incorrect
IP CAMERA login: root
Password:
# cd ..
# cd home
# pwd
/home
# ls
1080_mem_cfg.bin           cloud_oversea.ini          image.ini
HACKSD                     custom_init.sh             md5file
PATH                       debugTool                  ptz.cfg
VOICE-orig.tgz             drv                        rsyscall.goke
VOICE.tgz                  etc                        sensor.sh
VOICEON                    extra                      sensordetect
ap                         eye.conf                   sensors
busybox                    factory_tool.sh            start.sh
ca-bundle-add-closeli.crt  gk_fw.bin                  work.log
check_mem.sh               hardinfo.bin               wpa_supplicant.conf
chmemcfg                   hwcfg.ini
cloud.ini                  idx.log
#

gd-99, 2019-03-21:
@tonbor the output from your ls command is similar to mine, but I have some extra files as well. I don't know if it is possible to use other firmware.

I do not know if this is the best method of understanding if a partition is writeable, however using these commands:
`df -h | grep home`
I get: 
`/dev/mtdblock4            2.4M      2.1M    320.0K  87% /home`
Which is telling me my /home directory is 87% full, so some space left.
This command:
`mount | grep home`
I get: 
`/dev/mtdblock4 on /home type jffs2 (rw,relatime)`
My understanding of the (rw,realtime) is the home partition is rw - read writeable.
But doing:
`mount | grep /p2pcam`
I get:
`/dev/mtdblock3 on /p2pcam type squashfs (ro,relatime)`
Indicates /p2pcam is read-only. The df command also reports it is 100% full.

tonbor, 2019-03-21:
IP CAMERA login: root
Password:
# cd..
-sh: cd..: not found
# cd ..
# cd home
# df -h | grep home
/dev/mtdblock4            2.4M      2.4M     32.0K  99% /home
# mount | grep home
/dev/mtdblock4 on /home type jffs2 (rw,relatime)
# mount | grep /p2pcam
/dev/mtdblock3 on /p2pcam type squashfs (ro,relatime)

paus56, 2019-03-21:
у вас файл VOICE-orig.tgz  занял всю свободную память, удалите его как можно быстрее, тогда будет доступна запись в папку HOME

Google Translane:
you have a VOICE-orig.tgz file that takes up all the free memory, delete it as soon as possible, then you can write to the HOME folder

tonbor, 2019-03-21:
Great, that works!!!!!!!!!

tonbor, 2019-03-21:
Thanks paus56 and gd-99 for all the help

tonbor, 2019-03-22:
Got almost everything working. RTPS is on port 554 not on 8001 (anymore), login admin password passw0rd (self input).  Wifi is working, PTZ is back again, due to making space by deleting VOICE-orig.tgz.
Onvif is not working with the Onvif Device manager and OSD is 01-01-1970, I followed every step gd-99 wrote down. Any suggestions?

gd-99, 2019-03-22:
@tonbor, my cam can take a couple of minutes to pick up the correct time off the Internet. Once I replaced the ntp line in debug_cmd.sh with:
`(while true; do sleep 20 && /home/busybox/ntpd -q -p 0.uk.pool.ntp.org; done ) & `
the NTP mostly worked.

When it hasn't picked up the time, it has been down to a DNS issue. Assuming the DHCP server (router) is set up correctly, this should sort out DNS on its own. To check if DNS is your issue or not, log into the camera and do:
`ping google.com`
This should respond with something like:
PING google.com (**216.58.214.14**): 56 data bytes
64 bytes from **216.58.214.14**: seq=0 ttl=54 time=19.565 ms
64 bytes from **216.58.214.14**: seq=1 ttl=54 time=16.292 ms
...
This indicates DNS is working by resolving google.com to 216.58.214.14. If you do not get any IP indication then it is probably a DNS issue.

The DNS entries can be seen `with cat /etc/resolv.conf ` you should see something like:
nameserver  8.8.4.4
nameserver  8.8.8.8
These are google's DNS server addresses, yours are likely to be different. The first one may well be your routers address. If resolv.conf is empty you could try adding these entries manually using nano /vi, then do the ping test. Once you can successfully get a ping test to resolve a name to an IP address ntpd should just start working. Type `date` on the command line will respond with the full date and time of the camera.

If resolv.conf is the issue you may be able to get it automatically populated by adding:
`dns = 8.8.8.8 `to the `cls.conf` file. I haven't tried doing this without supplying all the other entries to cls.conf (as posted above) as my DHCP configuration just works.

tonbor, 2019-03-22:
Well ping was working Alright. I read all 21 Russian pages. I believe that
more people had the same problem, the date back to 1970 after a restart. I
believe the solution is to copy a hacked file from the russian site. Well I
cab not login there. More over copying files to the ip camera gives
possible bricking problems.

tonbor, 2019-03-25:
To connect to Synology Surveillance station 8.2 use user defined, stream
path: login:password@ipaddress:554/onvif/snapshot...... so  like ->
admin:passw0rd@192.168.1.100:554/onvif/snapshot

gd-99, 2019-03-25:
Thanks for the share. On my camera I have found a (netstat) diagnostic stream on port 3201. So http://<camera_IP>:3201 gives up a non ending stream of data from the camera. I have turned mine off by wrapping a file check  (if -f /somefile )around the "tees" line in the start.sh file. This way I can turn the diagnostic info off and on by the presence (or not) of "somefile" on the sdcard.

Some of the info coming out on port 3201 is "event: motion" and "event: sound". Should be possible to filter the "events" to kick of some sort of action on/from the camera - send mail maybe?

tonbor, 2019-03-31:
Well I managed to make a conf file for this camera  for the synology surveillance version 6 too. It is working there but not yet PTZ.

tonbor, 2019-04-02:
Well I got it working Tinycam with ptz, cheap common nvr  HZXM_NBD7808T with ptz, 920p only, Synology version6 and 8.2, not yet PTZ

tonbor, 2019-04-03:
OSD working ok now, only got UK summer time, I am 1 hour later. GMT+1 or something else like GMT+3 in start.sh does not change the OSD time

tonbor, 2019-04-04:
anyone got the daylight savings time problem tackled?

gd-99, 2019-04-04:
@tonbor I haven't had time to see if there is a way to properly do auto daylight saving. As a work around I can get my camera to match UK (I am in the UK) summer time. I will probably have to change it again when I go into winter time - if we change...

What I did: edit `/home/start.sh `and change the echo "whatever" > etc/TZ to
`echo "GMT-1" > /etc/TZ`

GMT-1 makes my camera clock (after ntpd has run) run one hour ahead of GMT. I understand - not tested - GMT+1 will make the clock run one hour behind GMT.

tonbor, 2019-04-04:
Well that is not working, GMT+1 or something else like GMT+3 in start.sh does not change the OSD time. See same problem like here http://4pda.ru/forum/index.php?showtopic=928641&view=findpost&p=81651102

BenDavidson90, 2019-05-18:
Hello, thank you for the hack and thank you @paus56 for the ONVIF activation setting.
My camera is now visible and controllable on ONVIF Device Manager (Windows) and ONVIFER (Andoid), however I can't add it to my Synology Surveillance Station...
It detects the ONVIF camera well and connects to it but cannot retrieve the image. However, if I use the RTSP stream it works but in this case no PTZ control...

Someone have a solution ?

evolucja, 2019-06-11:
Any way to get it working on Synology? On Android managed to have PTZ but not on Synology..

Jamyz, 2019-06-11:
It's possible yo use this with synology and PTZ ?

tonbor, 2019-06-11:
Well I got it working Tinycam with ptz, cheap common nvr HZXM_NBD7808T with
ptz, 920p only, Synology version6 and 8.2, not yet PTZ for Synology

aHVzY2g, 2019-07-30:
> 
> 
> Well I managed to make a conf file for this camera for the synology surveillance version 6 too. It is working there but not yet PTZ.

could you share your config file and give me a hint there I should copy it to? Thx