ant-thomas/zsgx1hacks #113: Bullet Cam, Switch Off White Night Light (PiR Activated)

colin-bremner, 2019-04-24:
Firstly thanks @ant-thomas/zsgx1hacks & contributors. I can confirm that readonlyhack-v0.1 is working on my "No-Name-Bullet-Outdoor-1080p-IP-Cam" NOT DTZ. Details at end. 

[TLDR]
I  want to switch off the night mode white light. So the camera only works in night-mode black & white (At Night B&W. Colour during the Daytime). 
Basicaly the ilumination from the white led's only attracts flying insects that trigger the sensor to record. I'm a novice but willing to experiment. 
# gio
--------------------------------
gio -di 49
gio -do 49 1
gio -s 49 0
gio -g 49
--------------------------------
#

FYI readonlyhack-v0.1 
I had to increase the sleep time on your  script to 30 for my ip-cam to get correct time from "UK ntp server". tbh I couldn't tell if changing the TZ made any difference. 

Set Time
sleep 20 && /home/busybox/ntpd -q -p 0.uk.pool.ntp.org 
to 
sleep 30 && /home/busybox/ntpd -q -p 0.uk.pool.ntp.org

MY IP-CAM DETAILS
localhost login: root
Password:   (cxlinux)
# ls /tmp | grep -F 3
3.4.0.0928
SDK10823
gc2033
#
# cat /proc/partitions
major minor  #blocks  name

  31        0        320 mtdblock0
  31        1       1664 mtdblock1
  31        2       1152 mtdblock2
  31        3       2560 mtdblock3
  31        4       2496 mtdblock4
 179        0   31166976 mmcblk0
 179        1   31162880 mmcblk0p1
#
# dmesg
[    0.000000] Booting Linux on physical CPU 0
[    0.000000] Linux version 3.4.43-gk (root@localhost.localdomain) (gcc version 4.6.1 (crosstool-NG 1.18.0) ) #71 PREEMPT Fri Nov 10 15:20:07 CST 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: VIPT aliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: Goke IPC Board
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] AHB: 0x90000000  0xf2000000  -- 0x1000000
[    0.000000] APB: 0xa0000000  0xf3000000  -- 0x1000000
[    0.000000] PPM: 0xc0000000  0xc0000000  -- 0x200000
[    0.000000] BSB: 0xc2600000  0xf5000000  -- 0x100000
[    0.000000] DSP: 0xc2700000  0xf6000000  -- 0x18f0000
[    0.000000] USR: 0xc3ff0000  0xfe000000  -- 0x10000
[    0.000000] hal version = 20160913
[    0.000000] On node 0 totalpages: 9216
[    0.000000] free_area_init_node: node 0, pgdat 8045b31c, node_mem_map 80493000
[    0.000000]   Normal zone: 72 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 9144 pages, LIFO batch:1
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 9144
[    0.000000] Kernel command line: console=ttySGK0,115200 mem=36M rootfstype=squashfs root=/dev/mtdblock2 init=linuxrc
mtdparts=gk_flash:320K(U),1664K(K),1152K(R),2560k(A),-(H)
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 36MB = 36MB total
[    0.000000] Memory: 31776k/31776k available, 5088k reserved, 0K highmem
[... 554 lines cut ...]
[   25.410000] =====gc2033 drv create at 20161117 ver:20170210====
[   25.420000] ===================================================
[   25.430000] [sensor_get_dev_id 276]find sensor id: 00002033
[   25.440000] sensor board reset...
[   26.740000] sensor board reset...
[   27.020000] win_height:0 win_width:0
[   27.030000] win_height:0 win_width:0
[   28.410000] ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   38.830000] wlan0: no IPv6 routers present
#
# lsmod
8188fu 932958 0 - Live 0x7f0ad000 (O)
gc2033_ex 11110 0 - Live 0x7f0a7000 (O)
sensor 155912 1 gc2033_ex, Live 0x7f07d000 (PO)
audio 8700 4 - Live 0x7f077000 (PO)
media 259569 7 sensor, Live 0x7f02f000 (PO)
hw_crypto 1948 1 media, Live 0x7f02b000 (PO)
hal 31555 1 media, Live 0x7f01f000 (PO)
exfat 88974 0 - Live 0x7f004000 (O)
gio 1587 1 - Live 0x7f000000 (O)
#

If I can add anything further plaes ask.

Best Regards

colin-bremner, 2019-04-27:
# hardinfo.bin

<DeviceInfo version="1.0">
<DeviceClass>0</DeviceClass>
<OemCode>0</OemCode>
<BoardType>1007</BoardType>
<FirmwareIdent>eyeplus_ipc_gk_005</FirmwareIdent>
<Manufacturer>EYEPLUS</Manufacturer>
<Model>GK7102</Model>
<WifiChip>RTL8188</WifiChip>
<GPIO>
<BoardReset>36_0x00000000_0x0_0</BoardReset>
<SpeakerCtrl>49_0x00000000_0x0_0</SpeakerCtrl>
<IrCut1B>32_0x00000000_0x0_1</IrCut1B>
<IrCut2B>52_0x00000000_0x0_1</IrCut2B>
<IrFeedback>0</IrFeedback>
<IrCtrl>31_0x00000000_0x0_1</IrCtrl>
<WhiteLight>30_0x00000000_0x0_1</WhiteLight>
</GPIO>

colin-bremner, 2019-04-27:
hardinfo.bin

`
<DeviceInfo version="1.0">
<DeviceClass>0</DeviceClass>
<OemCode>0</OemCode>
<BoardType>1007</BoardType>
<FirmwareIdent>eyeplus_ipc_gk_005</FirmwareIdent>
<Manufacturer>EYEPLUS</Manufacturer>
<Model>GK7102</Model>
<WifiChip>RTL8188</WifiChip>
<GPIO>
<BoardReset>36_0x00000000_0x0_0</BoardReset>
<SpeakerCtrl>49_0x00000000_0x0_0</SpeakerCtrl>
<IrCut1B>32_0x00000000_0x0_1</IrCut1B>
<IrCut2B>52_0x00000000_0x0_1</IrCut2B>
<IrFeedback>0</IrFeedback>
<IrCtrl>31_0x00000000_0x0_1</IrCtrl>
<WhiteLight>30_0x00000000_0x0_1</WhiteLight>
</GPIO>
`

closed 2019-04-27 by colin-bremner

RussH007, 2023-03-24:
Did you find a way to do this? I have a similar issue where the white lights come on at night every time motion is detected

RussH007, 2023-03-26:
hwcfg.ini support_doublelight = 1 (was 2)