ant-thomas/zsgx1hacks #115: Bought Silvercrest IP CAM S-K 1920 Version 1.1.1.13 Model WAPP-JS (LIDL) MCU Version Z1709250

pai68, 2019-05-07:
Hello
I have bought some of the cameras at Lidl Online in Germany, Unfortunately, I have not been able to somehow find the video stream so that I can use the camera with another software. The camera apparently uses the cloud and an app on the smartphone. The app "Silvercrest IP CAM S-K 1920" is available in the Playstore. So I can show a picture but I would like to use the IP CAM with another software directly on the PC.
The camera only responds to ping,
Telnet is not possible.
A Portscann did not result in open ports on the IP Cam.
An idea how to get further here or where to start with the research.

Per Ideström    Email pai68@gmx.de

Best Regards

Per

roema, 2019-06-02:
Did you test the hack with the SDCARD?

pai68, 2019-06-02:
HI I did test the hack with the SD card but did not workbest Regards  PerManuel Rösel <notifications@github.com> hat am 2. Juni 2019 um 17:57 geschrieben:  Did you test the hack with the SDCARD?

roema, 2019-06-03:
Okay. If the sdcard dump didnt work for you, so you can create  a custom rootfs. I can help you, but i need a dump or firmware image from the SPI flash. I can unpack and repack the image for you.

pai68, 2019-06-04:
Hi Manuel, that sound great. How to dump the firmware from the device ?best RegadsPer

roema, 2019-06-06:
Hi pai68!
I have send you an email with some instructions.
Today, i have analyzed a cheap cam from chineese [(unitoptek)](https://de.aliexpress.com/item/32830574232.html?spm=a2g0s.9042311.0.0.79834c4dVa2MCJ)

I have created a small python script to extract the partitions like uboot, kernel, home, etc, modify and repack the image. At the moment i am working on a script for the sdcard, to flash the home and root partition without flashrom tool.

This weekend i will publish the scripts on my githut side...

pai68, 2019-06-06:
Hi Manuel, that sounds good, I´m looking forward to test the script as soon it is finish. best RegardsPer ( pai68)Manuel Rösel <notifications@github.com> hat am 6. Juni 2019 um 22:50 geschrieben:  Hi pai68! I have send you an email with some instructions. Today, i have analyzed a cheap cam from chineese (unitoptek)I have created a small python script to extract the partitions like uboot, kernel, home, etc, modify and repack the image. At the moment i am working on a script for the sdcard, to flash the home and root partition without flashrom tool.This weekend i will publish the scripts on my githut side...

springfieldsmith, 2019-12-17:
Hi Roema. I have a camera with the same Model and MCU version referenced in this thread. I am also encountering the same problem. Would you be willing to send me the same instructions so that I can use it with another system? Thanks so much for your help:)
L

roema, 2019-12-17:
Hi Springfieldsmith!

Wich model do you have?

Most models have an SPI flash. It is very easy to read. If the SDCARD hack doesn't work, you can use a miniftdi module to read the flash.
There may be a firmware update for your camera online. It would also be possible to modify this.

Send me the name of the model, then i will check it.

springfieldsmith, 2019-12-17:
Hi there. Thank you so much for your help! The model is either WAPP-JS or
AWS53 depending on where you look. I didn’t try the sdcard hack yet because
I don’t know how. There *is* a firmware update for my camera which I have
not yet downloaded too.

roema, 2019-12-17:
Can you send me a link from the firmware file?
I cant find anything about WAPP-JS or AWS53..

springfieldsmith, 2019-12-17:
I am having trouble locating it, but I think you already solved this issue
for the other person on this thread... the title says “IP CAM Model WAPP-JS
MCU Version Z1709250” my camera is the same.
You previously said, “Hi pai68! I have send you an email with some
instructions. Today, i have analyzed a cheap cam from chineese (unitoptek)
 I have created a small python script to extract the partitions like uboot,
kernel, home, etc, modify and repack the image. At the moment i am working
on a script for the sdcard, to flash the home and root partition without
flashrom tool.  This weekend i will publish the scripts on my github
account”

Would you be willing to send me the same instructions?

Thanks so much
:)

springfieldsmith, 2019-12-17:
[image]
[image]

roema, 2019-12-17:
Okay, you can test the readonly hack with a small modification.
Append the following at the end of debug_cmd.sh

`/media/hack/busybox-armv6l nanddump -f /media/hack/BLOCK0 /dev/mtd0
/media/hack/busybox-armv6l nanddump -f /media/hack/BLOCK1 /dev/mtd1
/media/hack/busybox-armv6l nanddump -f /media/hack/BLOCK2 /dev/mtd2
/media/hack/busybox-armv6l nanddump -f /media/hack/BLOCK3 /dev/mtd3
/media/hack/busybox-armv6l nanddump -f /media/hack/BLOCK4 /dev/mtd4`

Insert the sdcard and boot the cam. It may take a while to create a DUMP from the mtd devices. After that, poweroff the cam an combine the files BLOCK0 to BLOCK4 with

`cat BLOCK0 BLOCK1 BLOCK2 BLOCK3 BLOCK4 > image.bin`

Now you have a full image from the cam. Binwalk should give some informatioan about the filesystem..

springfieldsmith, 2020-04-11:
> I think I found the firmware files! I am unsure how to modify them though.
> Would you be willing to help me? I wish to add this ip camera to my
> primary surveillance (lorex) system, but it doesn't come up in a device
> search.

I found the firmware files!
https://www.dropbox.com/sh/bchmsn552qfe4mo/AAAvl5LQTPmnmZ7vv-YmZVfxa?dl=0

springfieldsmith, 2020-04-11:
Would you be willing to help me modify the firmware so I can use the camera
with another system?

roema, 2020-04-26:
Yes. wich file is the correct one? There are many files on dropbox

springfieldsmith, 2020-04-26:
this one
https://www.dropbox.com/sh/iri0eoaor3388s1/AABDrkgPtOmSb2XofPNcuQA8a?dl=0

roema, 2020-04-26:
ok. thats are the original firmware files?
if so, its easy, you can unpack rootfs with

unsquashfs rootfs-cpio_master.squashfs.img

Now take a look at the files and make you changes to enable telnet or dropbear ssh server..

roema, 2020-04-26:
Take a look at the file check_new_fw.sh ...
The update uri is https://fw.omguard.com/download/IPCAM/CGAG/WAPP-CS/version.xml
The firmware file is https://fw.omguard.com/download/IPCAM/CGAG/WAPP-CS/GM8136-1.1.1.66.zip

roema, 2020-04-26:
Take a look at the file init, in rootfs, append the following

/gm/bin/busybox telnetd
then create new rootfs like
mksquashfs squashfs-root rootfs_new -comp xz -b 131072 -all-root

create new md5sum with
md5sum rootfs_new

and change the checksum in rootfs-cpio_master.squashfs.md5 to the new one

pai68, 2020-04-26:
Hallo Manuel
  
  
   
  
  
   hat sich was getan mit der IP Cam von Lidl ?
  
  
   
  
  
   Gruß Per

JodokusPindakaas, 2020-04-27:
Hi, fyi, i am also interested in getting a (rtsp) stream from this Lidl camera. Keep up the good work. Some background information regarding JSW.

JSW aka Secufirst.
https://www.jswpac.com

Head Office
3F-3, NO. 700, Zhongzheng Rd., Zhonghe District, New Taipei City, Taiwan

Manufacturing Center
NO. 138, Sanjiang lndustrial Park, Hengli Town, Dongguan City, Guangdong Province, China

http://www.secufirst.com.tw/

https://ch.marketscreener.com/JSW-PACIFIC-CORPORATION-20706834/unternehmen/

Contact in Europe (Netherlands)
https://www.secufirst.eu/
Ericssonstraat 2, 5121 ML, Rijen
The Netherlands
31 (0) 85 00 80 888 
(Mon to Friday 9:00 am - 5:00 pm) 

I’m from Holland, if needed i can call them.

Regards Frank

JodokusPindakaas, 2020-04-27:
SecuFirst Expert Official Product Expert:
Dutch forum:
http://diy-nl-nl.forum.ibood.com/products/202662-190527/ip-buitencamera-met-nachtzicht/

*****
Google translate:
Our cameras have a very high security level. Everything revolves around quality and IT security.

We want to keep hackers out at all costs.

This means that you will not find an IP address of our camera visible in your router. You can also not access the camera via an IP address. This is all shielded to ensure your safety. You can only access and install the camera with the Android or Apple app. In addition, we have a Windows PC program, UGRS player, with which you can also view the images live on your PC.

This is the reason that we use a program for the PC instead of an IP address which can be entered in the address bar.

It's all about your safety. We want you to be the only one who can access the images.
*****

Let's prove them wrong.

Regards Frank

JodokusPindakaas, 2020-04-27:
To be clear, it’s (also) referring to our cameras:
https://www.secufirst.eu/collections/draadloze-ip-cameras-zonder-verborgen-kosten-secufirst/products/draadloze-ip-beveiligingscamera-pan-tilt-outdoor-cam214

bitfarmer, 2020-04-27:
They are most likely using the Mac-address of the camera to find  them. _Remember  that the first 3 hex digits of a Mac-address are a vendor assigned number_ and they could well be using some other protocol than **IP** to  communicate between devices. To prove this you need to get Wireshark running on the same network and look at what is passing between the devices. _Beware that network switches might make it difficult to see traffic between other devices._ Regardless, such protocols do not route over the internet so they would have to encapsulate them in a tunnel of some sort to achieve that.

roema, 2020-04-27:
I have no camera from lidl to test it. But i can make a test rootfs. 

Take a look at the zip file. https://ros-it.ch/rootfs-cpio_master.squashfs.zip

Now i have the busybox telnet server enabled on boot. If everything okay, camera should open port 23. 

I am not responsible for any damage!!!

springfieldsmith, 2020-04-27:
Wow! Thank you so much! I will report back soon with the results.
—Also, no big deal if damage occurs to the camera; it’s practically
unusable in its current state... And I made the mistake of buying 2, so if
it doesn’t work we can try again :)

Cheers mate!

roema, 2020-04-28:
No Problem.
You can recover the camera if everthing goes wrong. On the mainboard, there should be a spi-flash.
With a simple ftdi-mini interface you can read and write the flash....

If port 23 is open, connect on it. User should be root with no password.
After that, we can take a look at the system configuration.

MadAEng, 2020-08-15:
I’m a bit lost on some of this. If I understand correctly, it’s not possible to use a silvercrest camera as a ‘normal’ IP camera out of the box, right?

I see some stuff here that suggests it’s possible to hack the firmware to change that, but has anyone succeeded? 
If so, could you help this n00b out with a step by step kind of guide?

OptimusGREEN, 2020-12-22:
Hi,

I have a fullDump.bin for my camera that I need to flash. Can someone tell me the command to flash it via ftdi connection?

MisterPingoo, 2021-01-20:
Hello guys, nobody answered @MadAEng , does it mean nobody succeeded ? Where did you go @springfieldsmith in your research ?

I am really interested as a friend of mine just gave me the lidl camera (SilverCrest S-K 1920, aka SecuFirst CAM 114.
I wanted to integrate it in my domoticz setup but unfortunately I encountered the same issues as you all. It is inaccessible.

I am just waiting my SD card to arrive to begin my tests...

My current firmware version seems to be (in the app)
1.1.1.13
but on this site, I found there was a 1.1.1.66 (!!) version:
https://fw.omguard.com/download/IPCAM/CGAG/WAPP-ES/version.xml
(changing WAPP-CS by WAPP-ES, because my cam is WAPP-ES)
WAPP-CS stands for... China and WAPP-ES for Europe ? ( I don't know, just saying...)

I hope we'll find a way to use these cams without their awful app :)

I'll let you informed

springfieldsmith, 2021-01-20:
Hi there.
Manuel Rösel made edits to the firmware files and gives instructions
earlier in this thread... but I became too busy with other projects and
haven’t had time to follow through to see if it worked. I will eventually
get around to it and let you know.
You could also follow the instructions he gave and see for yourself.

benkuly, 2021-02-06:
I have 5 IP CAM S-K 1920. The hacks from the README.MD doen't worked. The image given by @roema earlier in this thread seems to load (blinks for a minute), but telnet access is not granted.
What are the next steps from here. I have time and really like to help to make this hack even better and work for this cameras :slightly_smiling_face:

MHeuvel, 2021-02-27:
Any luck Benkuly? I have the same cam and tried it as well. Not getting very far and a disassemble of the cam proved too difficult.

benkuly, 2021-02-27:
sorry, no luck yet...

obel38, 2021-04-05:
Hi Roema,

Please, i have same issue (need a RTSP channel on Lidl camera), i oculd not open telnet port 23 on camera with SD Card formatted and your script installed on it.

My firmware version is WAPP-ES
Z1709250

Please any idea ???

Thanks for help

Regards

roema, 2021-04-05:
Hi @obel38 !
Ok, will take a look at the firmware.

I have downloaded GM8136-1.1.1.66.zip, is it the right for you cam?

roema, 2021-04-05:
There is a file like auto_script.sh. 

```
if [ "$factory_mode" == "1" ] || [ -f "/mnt/sd/telnetd" ]; then
    telnetd
fi
```
Please test to create a file named "telnetd"  on the sdcard and check port 23 on the cam.

MHeuvel, 2021-04-06:
> There is a file like auto_script.sh.
> Please test to create a file named "telnetd" on the sdcard and check port 23 on the cam.

Hi Roema,
Kewl! For me, this works. Just enter root as user login and you're in. It will be fun to explore what can be done now with this cam. :-)

MHeuvel, 2021-04-06:
I have been testing all morning now. The RTSP stream can be found on rtsp://[ip]/live/ch00_0
Can't get audio though. I can't find out why. I am seeing some settings in the auto_script.sh, but I did not have any success so far.
There are other interesting settings in the auto_script.sh and can invite others to go through them.

Should I be able to find how the audio stream can be added, I will post it here.

benkuly, 2021-04-06:
> Hi Roema,
> 
> Please, i have same issue (need a RTSP channel on Lidl camera), i oculd not open telnet port 23 on camera with SD Card formatted and your script installed on it.
> 
> My firmware version is WAPP-ES
> Z1709250
> 
> Please any idea ???
> 
> Thanks for help
> 
> Regards

I have the same, but can not access telnet after creating telnetd-file. Also have no access to the rtsp-stream, that @MHeuvel mentioned.

MHeuvel, 2021-04-06:
Hi @benkuly - you only need to create a SD-card with a file called telnetd - and be aware that you do not include any extension. During the bootup process, the SD-card will be checked for the existence of this file and will start the internal telnet daemon when found. That's it. If it doesn't work you might have a different camera.

benkuly, 2021-04-06:
Sorry, edited my message an hour ago. Telnet works, but not rtsp :)

MHeuvel, 2021-04-06:
> Sorry, edited my message an hour ago. Telnet works, but not rtsp :)

Then I misunderstood your message as well. Then please log in and start the service by entering **rtspd** - see if it pops up. Starting it like this will not daemonize the service and will output information like the used URLs.

benkuly, 2021-04-06:
> > Sorry, edited my message an hour ago. Telnet works, but not rtsp :)
> 
> Then I misunderstood your message as well. Then please log in and start the service by entering **rtspd** - see if it pops up. Starting it like this will not daemonize the service and will output information like the used URLs.

rtspd seems not be installed: `-sh: rtspd: not found`

roema, 2021-04-06:
Hi All!

Nice to hear that!
So, the next part ...

Search for the file **factory.xml**, should be in a folder named **default_factory_space**.

Test to change the **Model_name** value from something like **WAPP-CS** to **RTSP**.

You can find a line in auto_script.sh 

```
if [ $model == "RTSP" ]; then
    set_factory_network
    /mnt/mtd/rtspd
```
Now RTSP app should be start at bootup

obel38, 2021-04-06:
I can connect thru telnet !
 i have an error message when logged : "login: can't chdir to home directory '/root'"
but please, could you tell me what to type to look into folders ?

Thanks

Regards

benkuly, 2021-04-06:
> Now RTSP app should be start at bootup

I think that bricked the camera. No ping, telnet, rtsp or LED light. Press reset until LED goes on also does nothing.

roema, 2021-04-06:
Yes, please test 192.168.1.99 address.

**I would also like to state that all tests are at your own risk!**

It is also possible to set the following command at the beginning of the start-up script like (last line)

```
#! /bin/sh
VERSION=0.72
interface="eth0"
default_ip=10.0.0.26
factory_static_eth0_ip=192.168.1.99
wifi_test="0"
OBL_condition="/tmp/OBL_condition"
#Engineering mode
/mnt/mtd/rtspd // start rtspd
.....
```
These are just attempts, I don't have a camera to test.

benkuly, 2021-04-07:
That worked, thanks. Is there a way, that we could install the complete hack on this cam?

benkuly, 2021-04-08:
I found a very easy way, to make the cam an rtsp-cam:
Just create a file named `rtsp` on the sd-card. Then this part auf `auto_script.sh` will be executed:

```bash
if [ "$boot_mode" == "r" ] || [ -f "/mnt/sd/rtsp" ]; then   
                                                                                                 
#               set_factory_network                                    
        /gm/bin/isp_demon                                                           
        /mnt/mtd/rtspd -j                                       
        exit                                                         
    fi   
```
The url for the rtsp-stream then is: `rtsp://[ip-address]/live/ch00_1`. But audio does not work.

obel38, 2021-04-08:
Benkuly, thank you so much !!!!!!!!!

It works like a charm !!!

obel38, 2021-04-08:
Benkuly, i tried after testing to put comment lines as explained, but this is not working anymore even with removing them !!!

Looks like my cam is dead ...

benkuly, 2021-04-08:
Your cant try to reset the cam wir the reset button. Did you forget to comnent the `fi`?

obel38, 2021-04-08:
Yes i have created a blank SD with a new file inside, same !

Reset does not solve the problem

benkuly, 2021-04-08:
And what exactly is not working anymore? Does telnet work? Then you could copy the snipped in this thread here and we could help you.

dominikcgn, 2021-05-03:
Hello guys,

I have to jump into this topic. I have also this camera, but everytime when I try install it, with the silvercrest app, it says memory fault and it is not possible to finish the installation. I do not know which firmware, because I cannot even get into the camera menu. 
Unfortunately I have no coding skills at all, so I hardly understand what you are talking about. But maybe a firmware update could help? Or a reset? 
Thanks in advance!

PvdGulik, 2021-08-06:
Hello,

I have a Byron Smartwares DIC-23216. It also has no rtsp or other sort of stream. Did anyone manage to got it streaming?

Thanks!

damdam51, 2022-10-31:
Hi, I also have this camera. There's no Home assistant integration (direct or via the cloud).
Totally newbie in programming, I'm asking some help.

I get the rtsp flux, but is there any way (HACS integration, flashing another open firmware,...) to get the PTZ features, motion detector, like any good ONVIF camera ?

Thanks in advance

benkuly, 2022-10-31:
@damdam51 I use this camera in home assistant as simple RTSP camera wihtout the other features.

[15 more replies]