ant-thomas/zsgx1hacks #119: after installing  readonlyhack-v01.z  no service 554

theoneuij, 2019-05-24:
Starting Nmap 6.47 ( http://nmap.org ) at 2019-05-24 10:32 CEST
Nmap scan report for 192.168.1.26
Host is up (0.042s latency).
Not shown: 994 closed ports
PORT     STATE SERVICE      VERSION
21/tcp   open  ftp          BusyBox ftpd (D-Link DCS-932L IP-Cam camera)
| ftp-anon: Anonymous FTP login allowed (FTP code 230)
| total 33
| drwxrwxr-x    2 500      500            740 Aug  6  2018 bin
| drwxrwxrwt    4 root     root          3000 Jan  1 08:00 dev [NSE: writeable]
| drwxr-xr-x    4 500      500            364 Nov 30  2017 etc
| drwxr-xr-x    8 root     root             0 Jan  1 08:00 home
| -rwxrwxr-x    1 500      500            281 Nov  7  2017 init
| drwxr-xr-x    4 500      500            661 Apr 25  2017 lib
| lrwxrwxrwx    1 500      500             11 Aug  6  2018 linuxrc -> bin/busybox
| drwxr-xr-x    7 root     root         32768 Jan  1 08:00 media
| lrwxrwxrwx    1 500      500              8 Aug  6  2018 mnt -> /tmp/mnt
| drwxr-xr-x    2 500      500              3 Jan 22  2017 nfsroot
| drwxr-xr-x    2 500      500              3 Jan 22  2017 opt
| dr-xr-xr-x   58 root     root             0 Jan  1 08:00 proc
| drwxr-xr-x    2 500      500              3 Nov  7  2017 root
| lrwxrwxrwx    1 500      500              3 Aug  6  2018 run -> tmp
| drwxrwxr-x    2 500      500            253 Nov  7  2017 sbin
| dr-xr-xr-x   11 root     root             0 Jan  1 08:00 sys
| drwxrwxrwt    5 root     root           360 Jan  1 08:00 tmp [NSE: writeable]
| drwxrwxr-x    7 500      500             75 Nov  7  2017 usr
| lrwxrwxrwx    1 500      500              8 Aug  6  2018 var -> /tmp/var
|_Only 20 shown. Use --script-args ftp-anon.maxlist=-1 to see all.
|_ftp-bounce: bounce working!
22/tcp   open  ssh          Dropbear sshd 2016.74 (protocol 2.0)
|_ssh-hostkey:
23/tcp   open  telnet?
7103/tcp open  tcpwrapped
8001/tcp open  vcom-tunnel?
8080/tcp open  http         BusyBox httpd 1.13
|_http-methods: No Allow or Public header in OPTIONS response (status code 501)
|_http-title: 404 Not Found
2 services unrecognized despite returning data. If you know the service/version, please submit the following fingerprints at http://www.insecure.org/cgi-bin/servicefp-submit.cgi :
==============NEXT SERVICE FINGERPRINT (SUBMIT INDIVIDUALLY)==============
SF-Port23-TCP:V=6.47%I=7%D=5/24%Time=5CE7AC40%P=arm-unknown-linux-gnueabih
SF:f%r(NULL,20,"\xff\xfd\x01\xff\xfd\x1f\xff\xfb\x01\xff\xfb\x03\r\r\nIP\x
SF:20CAMERA\x20login:\x20")%r(GenericLines,46,"\xff\xfd\x01\xff\xfd\x1f\xf
SF:f\xfb\x01\xff\xfb\x03\r\r\nIP\x20CAMERA\x20login:\x20\r\n\r\nIP\x20CAME
SF:RA\x20login:\x20IP\x20CAMERA\x20login:\x20")%r(GetRequest,32,"\xff\xfd\
SF:x01\xff\xfd\x1f\xff\xfb\x01\xff\xfb\x03\r\r\nGET\x20/\x20HTTP/1\.0\r\n\
SF:r\nIP\x20CAMERA\x20login:\x20")%r(HTTPOptions,36,"\xff\xfd\x01\xff\xfd\
SF:x1f\xff\xfb\x01\xff\xfb\x03\r\r\nOPTIONS\x20/\x20HTTP/1\.0\r\n\r\nIP\x2
SF:0CAMERA\x20login:\x20")%r(RTSPRequest,40,"\xff\xfd\x01\xff\xfd\x1f\xff\
SF:xfb\x01\xff\xfb\x03\r\r\nIP\x20CAMERA\x20login:\x20OPTIONS\x20/\x20RTSP
SF:/1\.0\r\n\r\nPassword:\x20")%r(RPCCheck,19,"\xff\xfd\x01\xff\xfd\x1f\xf
SF:f\xfb\x01\xff\xfb\x03\r\r\n\x80\^@\^@\(r\xfe\^\]")%r(DNSVersionBindReq,
SF:53,"\xff\xfd\x01\xff\xfd\x1f\xff\xfb\x01\xff\xfb\x03\r\r\n\^@\^\^\^@\^F
SF:\^A\^@\^@\^A\^@\^@\^@\^@\^@\^@\^Gversionbind\^@\^@\^P\^@\^CIP\x20CAMERA
SF:\x20login:\x20")%r(DNSStatusRequest,3A,"\xff\xfd\x01\xff\xfd\x1f\xff\xf
SF:b\x01\xff\xfb\x03\r\r\n\^@\^L\^@\^@\^P\^@\^@\^@\^@\^@\^@\^@\^@IP\x20CAM
SF:ERA\x20login:\x20")%r(Help,26,"\xff\xfd\x01\xff\xfd\x1f\xff\xfb\x01\xff
SF:\xfb\x03\r\r\nHELP\r\nIP\x20CAMERA\x20login:\x20")%r(SSLSessionReq,E0,"
SF:\xff\xfd\x01\xff\xfd\x1f\xff\xfb\x01\xff\xfb\x03\r\r\n\^\x08\^C\^@\^@S\
SF:^A\^@\^@O\^C\^@\?G\xd7\xf7\xba,\xee\xea\xb2`~\xf3\^@\xfd\x82{\xb9\xd5\x
SF:96\xc8w\x9b\xe6\xc4\xdb<=\xdbo\xef\^Pn\^@\^@\(\^@\^\x08\^@\^@\r\n\^@f\^
SF:@\^E\^@\^@e\^@d\^@c\^@b\^@a\^@`\^@\x08\x20\x08\x08\x20\x08\x08\x20\x08\
SF:x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08
SF:\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x0
SF:8\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\^@\^R\r\n
SF:\^@\^@\x20\x20\x20\x20\^@\^T\^@\^@\^H\^@\^F\^@\^C\^AIP\x20CAMERA\x20log
SF:in:\x20")%r(Kerberos,12E,"\xff\xfd\x01\xff\xfd\x1f\xff\xfb\x01\xff\xfb\
SF:x03\r\r\n\^@\^@\^@qj\x81n0\x81k\xa1\^C\^B\^A\^E\xa2\^C\^B\^A\r\n\xa4\x8
SF:1\^0\\\xa0\^G\^C\^E\^@P\x80\^@\^P\xa2\^\[\^BNM\xa3\x08\x20\x08\x08\x20\
SF:x08\x08\x20\x080\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x2
SF:0\x08\xa0\^C\^B\^A\^@\xa1\^N0\^L\^\[\^Fkrbtgt\^\[\^BNM\xa5\^X\^O1970010
SF:1000000Z\xa7\^F\^B\^_\^\^\xb9\xd9\xa8\x08\x20\x08\x08\x20\x080\x08\x20\
SF:x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\^B\^A\^
SF:R\r\n\^B\^A\^B\^A\^B\^A\^P\^B\^A\x08\x20\x08\x08\x20\x08\x08\x20\x08\x0
SF:8\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x
SF:08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\x08\x20\x08\
SF:x08\x20\x08\x08\x20\x08\x08\x20\x08\^B\^A\^A\^B\^A\^C\^B\^A\^BIP\x20CAM
SF:ERA\x20login:\x20");
==============NEXT SERVICE FINGERPRINT (SUBMIT INDIVIDUALLY)==============
SF-Port8001-TCP:V=6.47%I=7%D=5/24%Time=5CE7AC45%P=arm-unknown-linux-gnueab
SF:ihf%r(GetRequest,89,"HTTP/1\.1\x20404\x20Not\x20Found\r\nServer:\x20TAS
SF:-Tech\x20IPCam\r\nDate:\x20Fri,\x201\x20Jan\x2070\x2000:10:36\x20GMT\r\
SF:nContent-Length:\x209\r\nCache-Control:\x20no-cache\r\n\r\nNot\x20Found
SF:")%r(FourOhFourRequest,89,"HTTP/1\.1\x20404\x20Not\x20Found\r\nServer:\
SF:x20TAS-Tech\x20IPCam\r\nDate:\x20Fri,\x201\x20Jan\x2070\x2000:10:41\x20
SF:GMT\r\nContent-Length:\x209\r\nCache-Control:\x20no-cache\r\n\r\nNot\x2
SF:0Found")%r(HTTPOptions,9C,"HTTP/1\.1\x20405\x20Method\x20Not\x20Allowed
SF:\r\nServer:\x20TAS-Tech\x20IPCam\r\nDate:\x20Fri,\x201\x20Jan\x2070\x20
SF:00:10:47\x20GMT\r\nContent-Length:\x2018\r\nCache-Control:\x20no-cache\
SF:r\n\r\nMethod\x20Not\x20Allowed")%r(RTSPRequest,87,"RTSP/1\.0\x20200\x2
SF:0OK\r\nCSeq:\x200\r\nServer:\x20TAS-Tech\x20Streaming\x20Server\x20V100
SF:R001\r\nPublic:\x20DESCRIBE,\x20SET_PARAMETER,\x20SETUP,\x20TEARDOWN,\x
SF:20PAUSE,\x20PLAY\r\n\r\n");
Service Info: OS: Linux; Device: webcam; CPE: cpe:/h:dlink:dcs-932l, cpe:/o:linux:linux_kernel

Service detection performed. Please report any incorrect results at http://nmap.org/submit/ .
Nmap done: 1 IP address (1 host up) scanned in 170.09 seconds

theoneuij, 2019-05-24:
# netstat -a
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 0.0.0.0:ftp             0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:7101            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:7103            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:8001            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:3201            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:6670            0.0.0.0:*               LISTEN
tcp        0     64 192.168.1.26:ssh        192.168.1.25:56393      ESTABLISHED

theoneuij, 2019-05-24:
likes operating normal, only no service 554. Can you help me?

closed 2019-05-24 by theoneuij

hr1179, 2019-12-22:
Have you solved this. I have the same issue with a Digoo DG-W01F.