ant-thomas/zsgx1hacks #124: GK710X  Linux  SDK   ver. 2.0.0

paus56, 2019-07-11:
Linux SDK for our cameras on the GK710X
- uBoot
- kernel
- Onvif
- RTSP
- PTZ
- WEB
- ...

[image]
[image]

**[Download the Linux SDK](https://my.pcloud.com/publink/show?code=XZCIo37ZHW7xalTadqV6fqPaPmozB5ysq4M7)**

alesimoes, 2019-07-22:
Thank you. where do you found that SDK?

aalm, 2019-11-26:
Thanks, was good enough for me to build my own kernel w/builtin support for ext2, ext3 and ext4.
I had setup root-on-NFS working before, and was really missing for this!

Now i still gotta figure out why i get "detect sflash fail" during boot + empty /proc/mtd with custom
kernel, and then i think it's good enough to replace the one provided by vendor on flash.

ps. you can use GK710X_LinuxSDK_v2.0.0/linux/kernel/3.4.43/scripts/extract-ikconfig for dumping
the kernel config from (iirc. likely) /dev/mtdblock2

biot, 2019-11-26:
@aalm how did you manage to build a working kernel with that SDK? Mine hangs at boot, both with the Goke GCC and a regular one. Which GCC did you use?

This would *really* help me! I'm trying to get a mainline kernel running on the GK7101, documented [here](https://biot.com/blog/), and all help is welcome!

Incidentally @paus56 I would very much like to know how you got that SDK, and is there any more? An older version (to match what's on the camera), or a datasheet...

aalm, 2019-11-26:
GCC from the SDK did work for me, i'm using debian(i686/32bit) in a vm as my build host.

At which point does it exactly hang? Does it print anything? How are you loading it from u-boot?

i also have ip001_linux.rar[0] in which i haven't yet ran diff over to v2 SDK,
but it does look very close to same, just stripped to include less than v2 SDK,
and different set of old .configs. iirc. i found the link to it from 4pda.ru forums.

[0] sha256: 7B42BE862CF720788F9B8643249DAE7DFB4767DB2F1A39CEBF6E926492FF520E

biot, 2019-11-26:
> GCC from the SDK did work for me, i'm using debian(i686/32bit) in a vm as my build host.

I tried natively, and in a debian trusty VM but also 64 bit. I'll try a 32-bit VM.

> At which point does it exactly hang? Does it print anything? How are you loading it from u-boot?

Just "Uncompressing Linux... done, booting the kernel.". That's the decompressor, using its own UART driver. No output from the actual kernel.

Loading it from U-Boot with "dhcp; bootm". Feeding it the stock kernel from the tftp server that way makes it boot, so the method is ok.

> i also have ip001_linux.rar[0] in which i haven't yet ran diff over to v2 SDK,
> but it does look very close to same, just stripped to include less than v2 SDK,
> and different set of old .configs. iirc. i found the link to it from 4pda.ru forums.

Found it, thanks. Building that kernel within the SDK environment results in a kernel which hangs after the decompressor, exactly like the SDK kernel.

aalm, 2019-11-26:
Did you try with a .config as is extracted from your original vendor kernel?

iirc. both SDKs were preconfigured for "GK710XS", so i guess you you should atleast switch to "GK710X",
if using the .config(s) those SDKs ship with.

oh, i only just now noticed the link to your blog, interesting :]

biot, 2019-11-26:
Yes, did a manual config to GK7101 earlier, and tried with the extracted config from the booting kernel. The result was the same.

alesimoes, 2019-11-26:
Hi Bert, did you have tested the sample applications in your build? I trying to make the sample applications work using the kernel that come in the camera but not working. 
I'm creating a p2pcloud for those cameras.

biot, 2019-11-27:
@alesimoes no, I'm only working on the kernel.

paus56, 2019-11-27:
> I'm creating a p2pcloud for those cameras.

лучше использовать портированную прошивку оттуда:  https://github.com/ant-thomas/zsgx1hacks/issues/129

alesimoes, 2019-11-27:
Hi @paus56, I could see your posts in 4pda.ru and thanks for share the SDK. I'm able to build and write my application in camera but it is incompatible with adi lib. I think this SDK have an adi lib different of the cameras that I'm working on. 

My cloud works with H264 baseline profile, but the firmware have the Main profile. do you know how can I solve this issue?

biot, 2019-11-28:
@aalm still no luck with a 32-bit VM, but that was always voodoo :)
Any chance I could get a custom kernel of yours to try? Also, can I have a copy of your VM (with any account of yours sanitized of course)?

aalm, 2019-11-28:
Sure, we can try something.
I could start by compiling with your .config, if you share it via ie. http://ix.io or similar pastebin.

biot, 2019-11-28:
.config extracted from vendor kernel on mtdblock2: https://paste.ubuntu.com/p/2CYG2R4sY4/
.config I made manually and tried with SDK build: https://paste.ubuntu.com/p/JQ3zZMkSCw/

aalm, 2019-11-28:
didn't figure out better way to share the zImage[0].
built the extracted .config(after make mrproper), and answered GK710X for the system(ARCH_),
and "y" for the rest of new config options(minor differences between your&sdk option names).

[0] https://github.com/aalm/gk710x-tools/tree/zImage_v1

biot, 2019-11-28:
Ok, that kernel also doesn't boot for me, hangs at the same point. Somehow my test platform doesn't like these... I do have some other GK710x gear, will try on those.

aalm, 2019-11-29:
I wonder if the v1 (or v1.1.2 w/e the previous SDK version was) would work for that SoC better.

fwiw., i found out that atleast for me, the ip001_linux.rar version is more correct one to use,
as it has entry for "XM25QH64" in sflash_devices_supported[](gk_sflash_v1_00.c).

biot, 2019-11-29:
This SDK 2.0 is definitely ahead of what my hardware runs, I've found some differences in hardware writes and even hw_ops fields. Definitely would like to get my hands on an earlier SDK version.

Which GK7101 hardware are you using by the way?

aalm, 2019-11-30:
The camera i have is actually with GK7102S.

biot, 2019-11-30:
Do you have a link to it? I might get one to try out.

aalm, 2019-12-24:
> Do you have a link to it? I might get one to try out.

Missed your Q earlier; no, and unfortunately i wasn't able to find the product again.

But I just found something i think you might find interesting:
https://github.com/RT-Thread/rt-thread/tree/master/bsp/gkipc/armv6

and there was more under ../bsp/gkipc/,
but with a quick look it seems like above dir contains most of machdep bootstrap.

I hope you will continue your blog, it got me into trying out ghidra... i already knew about
it, but oh wow, reversing something above the size of kernel... is actually fun with a tool
like ghidra!!

i got a script for ghidra allowing saving modified bytes under selection back into the elf,
which does work out-of-box for only raw/flat binaries.
made ghidra more powerful for reversing/modifying the userland "firmware",
but if someone will get things running with mainline, that might not be necessary in
the end.

i've found enough "worth adapting" srcs w/reasonable licenses to cover most
of the features these cameras have/could use..
Repo(or a fork of it) linked below might turn into a thing "hosting" such tools.
https://github.com/bolshevik/goke-GK7102-customizer

biot, 2019-12-24:
> But I just found something i think you might find interesting:
> https://github.com/RT-Thread/rt-thread/tree/master/bsp/gkipc/armv6
> 
> and there was more under ../bsp/gkipc/,
> but with a quick look it seems like above dir contains most of machdep bootstrap.

That whole gkipc directory is specifically a port of RT-Thread to the Goke 710x platform (gkipc = Goke IP camera). It's easy to build, but... doesn't work. No interrupts. I've also ported ChibiOS to the gk7101, as it's small/easy to control. No interrupts. Nothing works :-(

> i got a script for ghidra allowing saving modified bytes under selection back into the elf,
> which does work out-of-box for only raw/flat binaries.
> made ghidra more powerful for reversing/modifying [...]

Yeah, I don't see serious problems on the userspace end, or even in porting kernel drivers (it's just a lot of work). But there's no way to even get started until I have a basic mainline kernel running, and that's where I'm stuck.

aalm, 2020-01-07:
> Yeah, I don't see serious problems on the userspace end, or even in porting kernel drivers (it's just a lot of work). But there's no way to even get started until I have a basic mainline kernel running, and that's where I'm stuck.

my best guess atm. is, that you might be running into a situation possibly referred [here](http://lists.infradead.org/pipermail/linux-arm-kernel/2012-March/091407.html).
i mean, for using !"gk-sio" you'd want the "interrupt_may_hang=true" behaviour,
which apparently doesn't exist by that name in mainline anymore, oh well.
at least iirc. gk-sio appeared to have somewhat similar 'watchdog' for (these?) issues..

innerfirebg, 2020-01-10:
Hello Everyone,
Can someone help me to build own binaries compatible with this camera, I'm using this SDK but when I try to execute the file, I get segmentation failed.

maartenjd, 2020-04-12:
[https://github.com/aalm]
> Thanks, was good enough for me to build my own kernel w/builtin support for ext2, ext3 and ext4.
> I had setup root-on-NFS working before, and was really missing for this!
> 
**> Now i still gotta figure out why i get "detect sflash fail" during boot + empty /proc/mtd with custom**
> kernel, and then i think it's good enough to replace the one provided by vendor on flash.
> 
> ps. you can use GK710X_LinuxSDK_v2.0.0/linux/kernel/3.4.43/scripts/extract-ikconfig for dumping
> the kernel config from (iirc. likely) /dev/mtdblock2

[https://github.com/aalm]
I managed to build a kernel using the 2.0.0 SDK and the .config extracted from my mtd2 but have the same error _detect sflash fail_ and that is why the mtd-partitions are not initialized. Did you ever find out why this happens or manage to solve this issue?
Thanks

aalm, 2020-04-12:
> [https://github.com/aalm]
>
>
> **> Now i still gotta figure out why i get "detect sflash fail" during boot + empty /proc/mtd with custom**
>
>
> [https://github.com/aalm]
> I managed to build a kernel using [...]

sure, iirc. i took some sflash definitions from the ".rar sdk".
i believe i've "documented" these changes somewhere, possibly linked to from [here](https://github.com/bolshevik/goke-GK7102-customizer/issues).

aalm, 2020-04-12:
> as it has entry for "XM25QH64" in sflash_devices_supported.

hmm, didn't find more than the quote from above, but it should be enough
for grep to find the .c-file from the SDK that needs editing for your device.

maartenjd, 2020-04-13:
> > as it has entry for "XM25QH64" in sflash_devices_supported.
> 
> hmm, didn't find more than the quote from above, but it should be enough
> for grep to find the .c-file from the SDK that needs editing for your device.

Set CONFIG_MTD_M25P80=y in .config.

Or do `make menuconfig` and go to Device Drivers > MTD Support > Self Contained MTD Device Drivers > Support most SPI Flash Chips (AT26DF, M25P, W25X, ...) and enable.
[image]

kumarsanjeevdutt, 2020-09-09:
i am not able to find the PTZ folder in the SDK ,could you please help me where i can find that file and how do i add it to my demo application.
Thank You

csurf, 2021-06-07:
Link to SDK download appears to be dead. I get 'invalid link code' message. Can someone pls provide a working link to download the SDK?

csurf, 2021-06-08:
@paus56 can you please provide an updated download link for the 2.0.0 SDK?

AngeloGomes, 2022-01-05:
Hi Guys, I´m trying to fix my cam. It seems to be in bootloop. After opening the case, I saw a Goke GK7101 chip and now I am trying to catch any info as much as possible. I noticed the SDK link is not working anymore. Can someone post a new link or share the SDK again. Thank you all.

0zoon0, 2022-03-18:
Hi @paus56, could you please fix the download link? Thank you.

paus56, 2022-03-19:
Download SDK:  https://disk.yandex.ru/d/oJJyHwfsp7bnKA

sh4tteredd, 2022-04-04:
Hi you all, does anyone know if can I build rtsp/onvif for my GK7101S based cameras (netvue vigil 2)? thanks

i'm trying to use this sdk but idk how it works...

csurf, 2022-09-04:
Can someone please help me understand how to compile a complete firmware image that includes the u-boot bootloader? 

I downloaded the SDK, ran `make deploy` and then `make`. The build seems to complete without any errors, but I don't see when or where the firmware image is created & stored. What `make` commands should I be running, and where is the built image stored?

Also, I've run into an issue where the only output on the serial console is `UGKMC`. In my case, the SPI flash chip on my camera died, and I was never able to get a full EEPROM backup (using 'ch341a' programmer). So, I replaced the flash chip & have tried flashing different images that I found on the 4pda.ru forum, but none of the images have worked. I'm guessing that none of these images include the bootloader, and so the GK7102 is unable to boot. Has anyone run into something like this? 

Without a datasheet, it's extremely difficult to understand what's going on.

csurf, 2022-09-08:
@maartenjd 
> Or do `make menuconfig` and go to Device Drivers > MTD Support > Self Contained MTD Device Drivers > Support most SPI Flash Chips (AT26DF, M25P, W25X, ...) and enable. And _detect sflash fail_ is gone and mtd-partitions are properly created.

Where or how did you do this? I couldn't find any Makefiles in the SDK that supported any type of `make config` or `make menuconfig`

maartenjd, 2022-09-09:
@csurf
I gave up a long time ago so really don't remember anymore. When I have some time again, I will have a look. Sorry.

csurf, 2022-09-14:
@aalm can you please help me figure out how to compile u-boot for the GK7102? I'm struggling to figure out how to get a working bootloader built. My original SPI flash chip died, and I didn't have a backup, so I need to build u-boot from scratch & load the u-boot image onto my flash chip. I've tried multiple build options for u-boot, but the processor won't boot. I simply get 'UGKMC' output on the serial console (UART0).

Wedmer, 2022-09-17:
@csurf SDK creates uboot, kernel, and rootfs images. For the start you need to flash only uboot. Then get other parts via tftp and flash them via uboot.

jhony251, 2024-03-13:
Hi friends!
this thread is for hacking cameras based on GK7101?
i cant understand all you have said.

I have others cams based on GK7102 and copying some files in SDCard all start to works.
could be possible to do something similar?

Thanks everyone.

easycome1981, 2024-06-28:
Can someone help me with a .bin tab for my ST-291-2M-AI camera, it uses GK7102S sensor 2053
Thank you.

TWINKLY09, 2026-04-14:
Does someone have the sdk files because none of the above link work anymore ?