ant-thomas/zsgx1hacks #125: Problem with GD-SC02 (semi-bricked) Help!!

tlkmcorp, 2019-08-02:
Hello
My camera was briked by a mistake (I touch hwcfg.ini incorretly).
I loaded firmware Guudgo GD-SC03 (because i dont have the firmware for GD-SC02). The camera worked again but the rotating none. 

¿Anyone has the firmare for GD-SC02? (or compress the folder home (at least) for try change the files and test if works again the rotating)

Help please!!
Thanks!!

OptimusGREEN, 2020-12-20:
how did you load the firmware?

tlkmcorp, 2021-02-13:
Hello  OptimusGREEN 
With the sd slot and rename the file to firmware.bin and a file called "ruishi-xiaoxueren-hwcfg.ini" that contains:
**[config]
model = CloudCamera
main_bps = 768
sensor_position = 0
support_ptz = 1
ptz_mcu = 1
support_eth = 1
sound_detect = 1
ir_detect_type = 2
adc_setting_max = 700
adc_setting_min = -300
support_onvif = 1**

Then, the camera says that the update was sucesfull.
Other method is with a ch341A programer to load a bin in the XM25QH128A (a memory chip 16Mbit that the camera has).

alexanderpoilov, 2022-07-22:
Hi, please, may give me firmware sc-002? Thanks

Sshellv, 2023-12-15:
Hi! if you still need it, I have it

capitanos95, 2023-12-27:
> Hi! if you still need it, I have it

Hi, I could use this. Could you send it to me?

Sshellv, 2023-12-27:
552 5.3.4 Message size exceeds fixed limit

capitanos95, 2023-12-27:
> […](#)

Sorry but I can't read your message. Could you repost it?

Sshellv, 2023-12-27:
File too big.  Mail delivery server return my email.

capitanos95, 2023-12-27:
> File too big. Mail delivery server return my email.

Upload it to some hosting service and kindly put the link here

Sshellv, 2023-12-27:
https://fex.net/s/aznyafo

capitanos95, 2023-12-27:
> https://fex.net/s/aznyafo

Ok, this is the firmware for the GD-SC02 model? Can you tell me the procedure on how to update. A thousand thanks.

Sshellv, 2023-12-27:
https://fex.net/s/ovooxbn

Key: ovooxbn

capitanos95, 2023-12-27:
> https://fex.net/s/ovooxbn Key: ovooxbn 

Ok, this is the firmware for the GD-SC02 model? Can you tell me the procedure on how to update. A thousand thanks.

Sshellv, 2023-12-27:
I am programming a memory chip

capitanos95, 2023-12-27:
> I am programming a memory chip 

So now how should I flash it? Isn't it enough to put the file in the SD and start the IP cam?

Sshellv, 2023-12-27:
No.  you need a flash memory chip.