ant-thomas/zsgx1hacks #127: Bricked Digoo MYQ

Fietspomp86, 2019-08-19:
Hey guys, tried to change my rstp password with the firmware.bin and hwcfg.ini file.
But somewhere something went wrong and now it's not really like it should.
It sort of boots but this is the log from serial connection:

[digoolog.txt](https://github.com/ant-thomas/zsgx1hacks/files/3517343/digoolog.txt)

I have a firmware.bin from Digoo, how can I flash this using my serial to usb connector?
Maybe than I can erase flash and start over.

Please help :-)

Fietspomp86, 2019-08-22:
I think this can be closed.
Turned out my homefolder was empty.
Made a new firmware using tenvis_pack and the home folder from this git.
After that I got the "no valid ip" error, turned out wpa_supplicant didn't want to start for some reason.
I still don't know what happened but apparently the folder "/var/run" didn't exist and this wpa_supplicant didn't start which caused wifi to fail.
Got it back up now by adding this in the debug_cmd.sh
`mkdir /var/run && /home/wpa_supplicant -B -iwlan0 -Dwext -c /home/wpa_supplicant.conf -P /home/wpa_pid.txt &`

Now it's working again!
Thanks to all the issues and git's I managed myself. I'd still like a dump of the MYQ if someone has this to fully get it like original again.

Fietspomp86, 2019-08-24:
Ok I'm closing this, device is stable for a couple of days already.
Updated my p2pcam with a version I found on 4pda.ru, works great now.
Still using the wpa_supplicant workaround but no problems there.

closed 2019-08-24 by Fietspomp86

OptimusGREEN, 2020-12-20:
my home folder is empty too and camera doesn't seem to boot. how did you re populate the home folder?