ant-thomas/zsgx1hacks #128: help with DIGOO DG-QB01

rdpeake, 2019-09-03:
Hey i have a DIGOO DG-QB01: [banggood](https://www.banggood.com/DIGOO-DG-QB01-Mini-Flexible-720P-2_1mm-Wireless-WIFI-Smart-Home-IP-Camera-Baby-Monitor-p-1259076.html?rmmds=search&ID=224&cur_warehouse=CN)

I have cracked it open enough to see it has a hi3518 chip inside and that i can access the serial port - which results in the following u-boot output:

```
SPI Nor total size: 16MB
PPS:Jul  4 2018 20:28:01   meari_c2:  0
magic err
ARM Linux Kernel Image (uncompressed)
2617352 Bytes = 2.5 MiB
   Loading Kernel Image ... magic err
Uncompressing Linux... done, booting the kernel.
```

I'm mostly a noob when it comes to this stuff - but can often follow instructions once found. My hope is to be able to hack this to enable a stream, and snapshot url to use with octopi, but i'm currently lost of where to go next in order to gain access to the linux system, or know if it is even possible. Any help you can give would be great

Fietspomp86, 2019-09-06:
Did you setup WiFi etc? If so, try to connect with telnet, there's a list with common/known passwords on GitHub, see if you can gain access to it and tinker on from there.

rdpeake, 2019-09-07:
Thanks fietspomp for the reply - unless i am doing something wrong i am incapable of accessing a telnet server (unless it is on a non-standard port). Trying to access the ip address via a browser shows up with a BASIC auth prompt but i couldn't see anywhere to find possible usernames and passwords for those online.

dgomes, 2020-01-28:
any progress on this ?

netomx, 2020-11-16:
It is trying to boot a corrupted kernel. Did you modify the kernel?

Clarkeeeee, 2021-07-07:
> Thanks fietspomp for the reply - unless i am doing something wrong i am incapable of accessing a telnet server (unless it is on a non-standard port). Trying to access the ip address via a browser [...]

Did you end up cracking the username and password? I have the same camera that was using the CloudEdge app that has now stopped working and rendered the camera useless. If I can somehow enable ONVIF I can access the stream and re-use the camera. 

Any help would be greatly appreciated