ant-thomas/zsgx1hacks #129: Wanscam K* firmware on INQMEGA HIP291

slydiman, 2019-09-28:
FYI
We have just released the patch of the latest Wanscam K21 firmware 16.06.00.52 to use it on the camera INQMEGA IL-HIP291G-2M-AI with the sensor gc2033.
https://4pda.ru/forum/index.php?showtopic=928641&st=900#entry89274008
Note the description is Russian.
We have patched few drivers to work on GK7102 instead of GK7102S and gpio pins map inside the main app.
It seems that the firmware Wanscam K series has been developed by ZSCam company. It works with SAP HD app and clones (YSee, WISE VR, Eyedot). It has very good and stable Onvif, RTSP and Web UI implementation. We have added telnet, ftp and other tools. We can help to adapt this firmware for other GK7102 cameras.

paus56, 2019-10-05:
[image]
[image]

svastarija, 2019-10-05:
Hi, the download link does work on 4pda.ru, it give 404 error

paus56, 2019-10-06:
To download files from 4pda.ru you need to register and login to your account

I have attached the firmware mod file here...     description of the installation to read on 4pda

[HIP291G_Mod_ENG_04102019.ZIP](https://github.com/ant-thomas/zsgx1hacks/files/3694126/HIP291G_Mod_ENG_04102019.ZIP)

nishant2019, 2019-10-06:
Where can I get INQMEGA IL-HIP291G-1M-AI firmware ?

svastarija, 2019-10-08:
> 
> 
> To download files from 4pda.ru you need to register and login to your account
> 
> I have attached the firmware mod file here... description of the installation to read on 4pda
> 
> [HIP291G_Mod_ENG_04102019.ZIP](https://github.com/ant-thomas/zsgx1hacks/files/3694126/HIP291G_Mod_ENG_04102019.ZIP)

Thanks..  greatly appreciated. I have tried to create an account on 4PDA but it is very hard for non RUS, since CAPTCHA is asking to add numbers in russian, making it very hard to gues.. anyway.. thanks

nishant2019, 2019-10-09:
> Thanks.. greatly appreciated. I have tried to create an account on 4PDA but it is very hard for non RUS, since CAPTCHA is asking to add numbers in russian, making it very hard to gues.. anyway.. [...]

Try this website
https://doorsgeek.blogspot.com/2015/08/4pdaru-loginregister-captcha-tutorial.html

paus56, 2019-10-14:
Обновление MOD прошивки от 20191014

Обновил mod прошивку для камер HIP291G
- добавил поддержку сенсора **gc1034** ( камера HIP291G-1M-AI )
- добавил поддержку сенсора **gc2053** ( камера HIP291G-2M-AI )

подробнее на 4PDA...

скачать прошивку также можно там: https://my.pcloud.com/publink/show?code=XZhab4kZdjDDCXQAGsHsODwRRr8reFq23KeX

nishant2019, 2019-10-15:
> 
> 
> Обновление MOD прошивки от 20191014
> 
> Обновил mod прошивку для камер HIP291G
> 
>     * добавил поддержку сенсора **gc1034** ( камера HIP291G-1M-AI )
> 
>     * добавил поддержку сенсора **gc2053** ( камера HIP291G-2M-AI )
> 
> 
> подробнее на 4PDA...
> 
> скачать прошивку также можно там: [HIP291G_Mod_ENG_20191014](https://my.pcloud.com/publink/show?code=XZhab4kZdjDDCXQAGsHsODwRRr8reFq23KeX)

как можно прошить мою камеру этой прошивкой? Можно ли использовать Uboot для этого?

paus56, 2019-10-15:
>> как можно прошить мою камеру этой прошивкой? Можно ли использовать Uboot для этого?

- можно использовать U-boot
- также можно просто скопировать 3 файла из архива в корень SD карты.

nishant2019, 2019-10-15:
> * можно использовать U-boot
> 
> * также можно просто скопировать 3 файла из архива в корень SD карты.

получать это сообщение
Mod FW programming
eyeplus_ipc_gks_001
Unknown target.

nishant2019, 2019-10-15:
Will this work on GK7102S based  IL-HIP291G-1M-AI gc1034

paus56, 2019-10-15:
no  work on GK7102S

nishant2019, 2019-10-18:
> 
> 
> no work on GK7102S

any firmware for GK7102S

rufussckingston, 2019-10-18:
I'm trying to D/L the Backup/restore zip files from the links in post 93, but am getting 404 errors... Can you post them here please.

THX

paus56, 2019-10-19:
[Backup.zip](https://github.com/ant-thomas/zsgx1hacks/files/3746283/Backup.zip)
[restore.zip](https://github.com/ant-thomas/zsgx1hacks/files/3746284/restore.zip)

Keeze, 2019-11-02:
Firmware also works on a INQMega PTZ381 https://www.banggood.com/2019-NEW-INQMEGA-PTZ381-HD-1080P-PTZ-360-Panoranic-Waterproof-IP-Camera-IR-Night-Version-Two-way-Audio-p-1542643.html?cur_warehouse=CN

[image]

Thank!

kirelam, 2019-11-02:
I wish there was modded firmware for the  gk7102S version

paus56, 2019-11-03:
> Firmware also works on a INQMega PTZ381 
>
This firmware supports all the functionality of the INQMega PTZ381 camera? 
PTZ works correctly with this firmware ?

Keeze, 2019-11-03:
Camera seems to fully support PTZ, had to flip the controls though (this cam hangs from a mount). Sensor is 
```
# ls /tmp | grep -F gc
gc2033

```

paus56, 2019-11-04:
Собрал прошивку для камер на процессоре **GK7102S**
скачать можно там: 
[Hip_Mod_20191103_ENG_7102S](https://my.pcloud.com/publink/show?code=XZdXkQkZFts5BUzYpcmzln4nkSQx7yD4Smek)

подробности на [4PDA](http://4pda.ru/forum/index.php?s=&showtopic=928641&view=findpost&p=89274008)

paus56, 2019-11-27:
!!!!!  Возможны проблемы с мод прошивкой для камер на процессоре GK7102S
подробнее читать там:  http://4pda.ru/forum/index.php?s=&showtopic=928641&view=findpost&p=91290366

aalm, 2019-11-30:
Hi,

could you share a log of "ipc" working correctly via console(=uart)?
(of course, do censor if there's any UIDs/private info in it), thanks in advance:]

i have a camera with GK7102S(exactly like in [this pic](https://www.mikrocontroller.net/attachment/357761/IMG_6787.JPG)),
and currently running the ipc from Hip_Mod_20191103_ENG_7102S does cause
reboot almost instantly.

apparently it might atleast support the sensor(sc1135) my camera has:
`# strings /home/ipc | grep -C9 sc1135`
`ipc.c`
`sensortype=%s`
`[%d %s,%s, %d ]=>ret=%d, val:%s`
`sc1145`
`[%d %s,%s, %d ]=>sc1145`
`sc2135`
`[%d %s,%s, %d ]=>sc2135`
`ov9750`
`[%d %s,%s, %d ]=>ov9750`
`sc1135`
`[%d %s,%s, %d ]=>sc1135`
`sc2315`
`[%d %s,%s, %d ]=>sc2315`
`sc2232`
`[%d %s,%s, %d ]=>sc2232`
`sc2235`
`[%d %s,%s, %d ]=>sc2235`
`jxf23`
`[%d %s,%s, %d ]=>jxf23`
`/tmp/.sensortype.ini`

i guess i should begin reverse-engineering ipc, to figure out how to 'satisfy'
the cause of reboot, since there was no obviously fatal errors visible via console..
the camera came with p2pcam originally, but i'd like to replace that.

paus56, 2019-11-30:
вот лог загрузки камеры на процессоре **GK7102**

[20191130_mod_out.zip](https://github.com/ant-thomas/zsgx1hacks/files/3907219/20191130_mod_out.zip)

aalm, 2019-11-30:
@paus56 Thank you for the log & all the info you've shared about these cams :]

fwiw., i began taking a closer look at what "ipc" does, and i built [strace(1)](http://man7.org/linux/man-pages/man1/strace.1.html) for that, which
i wasn't able to find else where.

this is the last version that was on sourceforge from iirc. 2018,
but i will try to get around to building a newer version from the official source later:
[static-strace-4.21.zip](https://github.com/ant-thomas/zsgx1hacks/files/3907519/static-strace-4.21.zip)

about the failure i'm having, i'm now wondering if it's related to this:
`04:08:09.418298 [76e8b384] open("/proc/mounts", O_RDONLY) = 12</proc/179/mounts>`
`04:08:09.421802 [76e8b4a4] read(12</proc/179/mounts>, "rootfs / rootfs rw 0 0\n/dev/root / ext2 rw,relatime,errors=continue 0 0\nproc /proc proc rw,relatime 0 0\ntmpfs /dev tmpfs rw,rela"..., 4096) = 483`

so "ipc" is obviously looking at the mounts,
and might not like seeing squashfs missing and/or ext2 there?

having rootfs/storage on anything but the 8M spi flash would be ideal for testing/hacking on
these "firmwares".

Anon0ne, 2019-12-18:
I tried this on the camera INQMEGA PTZ but it must have bricked it. Once I loaded the firmware on SD card and restart it.. I restart device in 3-5 minutes and it never come back online.

Now I have tried to connect to UART and access UBoot but I  am getting no output at all from it.. did the flash chip get killed? wtf.

Anon0ne, 2020-01-14:
Got a replacement today, and I dumped the mtd0 throught mtd4 and I cannot find a way to download the merge tool to share this firmware.

paus56, 2020-01-14:
теперь мод Wanscam K21 firmware может работать на этих камерах  с SD карты без прошивки флеш...
файлы и описание подробно там: http://4pda.ru/forum/index.php?s=&showtopic=928641&view=findpost&p=92865876

aalm, 2020-01-16:
@paus56 i saw this was missing, works for me:]
[edit_config.arm.zip](https://github.com/ant-thomas/zsgx1hacks/files/4070170/edit_config.arm.zip)

Anon0ne, 2020-01-16:
I need the edit devParam.dat and image merge tool from 4pda. The FTP file would help too but I cannot make am account there for anything.

Anon0ne, 2020-01-16:
> Firmware seems to fully support all features, including PTZ, had to flip the controls though (this cam hangs from a mount). Sensor is
> 
> ```
> # ls /tmp | grep -F gc
> gc2033
> ```

Careful I got a PTZ381 recently that did not take the firmware very well. In fact the flash chip quit responding all together. I will have to desolder old one to burn in CH341A.

There are ways to mod the new one with the zxsg hacks manually and safely.

aalm, 2020-01-29:
> We have patched few drivers to work on GK7102 instead of GK7102S and gpio pins map inside the main app.
> It seems that the firmware Wanscam K series has been developed by ZSCam company. It works with [...]

@slydiman, @paus56, if you are familiar with C, and have been reversing the 'ipc',
you might find my project interesting. In the end(once it's usable for rtsp etc.)
the result should be much better than changing bytes in 'ipc' manually, i hope:]
you can find the Work-In-Progress project from: https://github.com/aalm/gkipc_hooked

slazaris, 2020-02-13:
Hello, is this firmware also compatible with IL-HIP291G-1M-AI, the 1MP version of the camera. Is it possible to flash back the original firmware in case something doesn't work. Can you also share a link with the original firmwares?

yomismo112, 2020-02-20:
Hi. I've bricked a Inqmega IL-HIP329G-1-C. I've tried flashing various versions of software but none has run. Is there a version of this software (HIP291G_Mod_ENG_04102019.ZIP) that i can flash directly to the eeprom? Thanks in advance.

Anon0ne, 2020-02-20:
> Hi. I've bricked a Inqmega IL-HIP329G-1-C. I've tried flashing various versions of software but none has run. Is there a version of this software (HIP291G_Mod_ENG_04102019.ZIP) that i can flash directly to the eeprom? Thanks in advance.

I did the same thing on the first one I got. I was unable to find anything to flash to it, I got the programmer but had no luck with the chip still on board. I read on 4pad.ru you have to remove it to flash it.

And you probably needed to have your own backup. I have a firmware backup but who knows if it works for others.

yomismo112, 2020-02-23:
Hello AnonOne. I've unbricked the camera at last. I explain you how:
-With a programmer i extract the firmware from another camera (with the same processor and image sensor) that i own.
-Then with this firmware i flash the old camera and the come back to life.
-I've attached the firmware for if you want to probe.
-When you flash the firmware it's important desolder the pin number 8 of the flash memory chip and select the correct type of flash memory in the programmer software (i use an CH341A).
If you need help don´t doubt in ask me that you need.
Sorry for my bad english, i'm spanish !!!!

[firmware.zip](https://github.com/ant-thomas/zsgx1hacks/files/4242509/firmware.zip)

maartenjd, 2020-03-22:
> To download files from 4pda.ru you need to register and login to your account
> 
> I have attached the firmware mod file here... description of the installation to read on 4pda
> 
> [HIP291G_Mod_ENG_04102019.ZIP](https://github.com/ant-thomas/zsgx1hacks/files/3694126/HIP291G_Mod_ENG_04102019.ZIP)

Hello paus56,
I would like to put this firmware on my camera. When I look at the debug_cmd.sh file that you provided, this firmware needs 6 mtd-partitions:
_if [ -f $MNTPT/$FLASHBIN ];then
	dd if=/dev/mtd0 of=$MNTPT/mtd0.im
	dd if=/dev/mtd1 of=$MNTPT/mtd1.im
	dd if=/dev/mtd2 of=$MNTPT/mtd2.im
	dd if=/dev/mtd3 of=$MNTPT/mtd3.im
	dd if=/dev/mtd4 of=$MNTPT/mtd4.im
	dd if=/dev/mtd5 of=$MNTPT/mtd5.im_

My camera now has 5:
_mtdparts=gk_flash:320K(U),1664K(K),1152K(R),2560K(A),-(H)_

what gives
_[    0.640000] 0x000000000000-0x000000050000 : "U"
[    0.650000] 0x000000050000-0x0000001f0000 : "K"
[    0.660000] 0x0000001f0000-0x000000310000 : "R"
[    0.670000] 0x000000310000-0x000000590000 : "A"
[    0.680000] 0x000000590000-0x000000800000 : "H"_

I cannot install the HIP291G_Mod_ENG_04102019-firmware because it cannot find the /dev/mtd5. Are you sure that it needs 6 mtd-partitions? In that case I would like to see what is in the mtdparts-command line in the u-boot environment. Can you provide that?

Strangely, when I look in your 20191130_mod.log, I also see only 5 mtd-partitions:
_[    0.640000] 5 cmdlinepart partitions found on MTD device gk_flash
[    0.650000] Creating 5 MTD partitions on "gk_flash":
[    0.650000] 0x000000000000-0x000000050000 : "U"
[    0.660000] 0x000000050000-0x0000001f0000 : "K"
[    0.670000] 0x0000001f0000-0x0000003c0000 : "R"
[    0.670000] 0x0000003c0000-0x000000410000 : "Cg"
[    0.680000] 0x000000410000-0x000000800000 : "H"_
so I begin to doubt if your debug_cmd.sh is correct? When I binwalk the firmware, I see the familiar addresses show up: 0x50000, 0x1f0000, 0x3c0000, 0x3c0000 and 0x410000, just as in your log file.

Just delete the mtd5-line from the script?

Or I could just manually flash the mtd1-4 partitions. I will see if that works.

paus56, 2020-03-23:
в файле debug_cmd.sh находится этот скрипт:

для создания бэкапа вашей оригинальной прошивки с количеством mtdX до 6...  при этом реально блоков mtd может быть сколько угодно, но не больше 6...
если их меньше, то они будут проигнорированы...

!!! я вам рекомендую НЕ ПРОШИВАТЬ камеру, а использовать прошивку, которая работает с SD карты без записи флеш...

kamilmirza, 2020-03-25:
@slydiman @paus56 
flashed it on INQMEGA HIP291-1M-AI, works perfectly through web
any method to make it work on tinycam android app?

kamilmirza, 2020-04-04:
@slydiman @paus56 
can we disable onvif or secure our video feed to use it as a indoor camera, like we could do it on original patched firmware of INQMEGA HIP291

lolomin, 2020-09-22:
Hi,

Do you known if this firmware could work on Wanscam K64A please ?
The original firmware is not stable, camera reboots often, RTSP is not secured by login/pass, Home position not available, Presets  are not working, OnVif Left/Righ/Up/Down are inverted, no FTP Upload capabilities, max 15 fps ... big big regression comparing to HW* series from Wanscam.

Regards,

lolomin

AivanRam, 2020-10-04:
> Обновление MOD прошивки от 20191014
>
> Обновил mod прошивку для камер HIP291G
>
>     * добавил поддержку сенсора **gc1034** ( камера HIP291G-1M-AI )
>
>     * добавил поддержку сенсора **gc2053** ( камера [...]

Hi
Amazing update, I just tested on a Y4A-WB, same processor than INQMEGA HIP291. It works perfectly, I have just some questions.
>I am using Ysee to connect to my camera, there is updates on the firmware, is better not to touch this Updates?
>There is a way to save images and videos on ftp or free cloud, like onedrive, gdrive, etc?

It looks like the user has a bug, you can create users, it does not show them, but they works, also the device admin user crashes when you change the admin name "admin" for another.

Thanks!

svastarija, 2021-02-19:
Could anyone post link to  Hip_Mod_20191103_ENG_7102S ,all links are dead. thanks

Cryinman82, 2021-11-12:
> Firmware also works on a INQMega PTZ381 https://www.banggood.com/2019-NEW-INQMEGA-PTZ381-HD-1080P-PTZ-360-Panoranic-Waterproof-IP-Camera-IR-Night-Version-Two-way-Audio-p-1542643.html?cur_warehouse=CN
> 
> [image]
> 
> Thanks!

Hi!
Is it possible to have the link of the good firmware who work? And if possible the how to?
Thanks you by advance!

Cryinman82, 2021-12-03:
> Firmware also works on a INQMega PTZ381 https://www.banggood.com/2019-NEW-INQMEGA-PTZ381-HD-1080P-PTZ-360-Panoranic-Waterproof-IP-Camera-IR-Night-Version-Two-way-Audio-p-1542643.html?cur_warehouse=CN
> 
> [image]
> 
> Thanks!

Can you tell me which firmware you used? And the instructions link? Please!!!

csurf, 2022-09-15:
@yomismo112 
I have the same problem as you. I'm looking for a firmware bin file for my camera. My camera has the GK7102C chip, and un unknown image sensor. What are the specifications of your camera? I'm looking for any full eeprom image for the GK7102C that will at least allow me to reinstall U-Boot bootloader. I have tried flashing many images that I found here on github and on 4pda, but none of them have worked. All I see on the UART port is 'UGKMC'.

Wedmer, 2022-09-17:
@csurf, can you provide photo of the board with readable labels on chip?

maartenjd, 2022-09-17:
@csurf : I have put some FW-images (full and per partition) over here: [https://www.dropbox.com/sh/buza1pkjqgbtcg5/AAA-Vcazu_2AYxdzk2GnHAxha?dl=0](url). It is the dump of an 8 MB image.
_0.bin_ should be the u-boot image.

There is also the file fw0(HIP291G_Mod).img that I extracted from a HIP291G_Mod-fw-image.

As you have no serial access you will have to flash directly on the BIOS-chip.

csurf, 2022-09-18:
@maartenjd thanks, but I'm fairly certain that those images won't work, since your camera is a GK7102S. I'm pretty sure that the GK7102C usese a different build that isn't compatible with either the GK7102 or GK7102S. It would be cool if someone who has access to a camera with a GK7102C and another variant could try uploading a different firmware to the GK7102C and see if it works. 

At this point, I'm starting to wonder if my chip might be damaged since none of the firmware images I've been able to find online have worked. However, none of those images have been specifically for the GK7102C, so I still have some hope that I simply need the right image.

@Wedmer see pics
[image]
[image]
[image]

Wedmer, 2022-09-18:
@csurf I've found firmware for GK7102C devices https://sv3c.com/fireware/

csurf, 2022-09-19:
@Wedmer 
interesting. I also previously found that same link, but the firmware appears to be in a special, proprietary package. I believe you must use their software in order to upgrade the firmware. I simply need the raw binary firmware image to flash to the SPI chip, but I do not know how to extract the pure firmware from those pkg / zip files.

paus56, 2022-09-19:
[@csurf](https://github.com/csurf) 
у вашей камеры: **GK7102C** +  **F37** CMOS sensor 
у меня есть две прошивки для GK7102C  но в них нет поддержки F37 сенсора..
одна прошивка для флешь 8 MB,  другая для флешь 16 MB 
какой размер вашей флешь памяти в камере ?

maartenjd, 2022-09-19:
I downloaded GK7102C-AHX_V13.1.41.15.3-20200414(1080P.8M).pkg and did a first `binwalk` on it. That resulted in:

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
512           0x200           Squashfs filesystem, little endian, version 4.0, compression:xz, size: 2655800 bytes, 280 inodes, blocksize: 262144 bytes, created: 2020-04-11 09:14:24

So, you could try to mount the pkg as a squashfs in linux and then see what is in it.

easycome1981, 2024-06-28:
Can someone help me with a .bin tab for my ST-291-2M-AI camera, it uses GK7102S sensor 2053
Thank you.