ant-thomas/zsgx1hacks #133: Newer Iegeek (I think ?) gk7102 camera doesn’t boot from sd

Nickedemuz, 2019-10-23:
Hi

I recently got a IP camera that I think is an iegeek. Not sure though, the eBay ad just said “Ip WiFi cam” :)
[image]

I can’t get the hack to work, neither the one for older cameras nor the one for newer. I hooked it up to a serial connection so I can view the boot process via teraterm. I can also stop the boot by pressing enter and then get into U-boot. 
[image]

However, in U-boot I don’t know how to get it to boot a clean Linux of any kind, even less this hacked firmware. I can type “boot” in u-boot and get it to boot the original firmware but as probably everyone reading this I don’t want that ;)
Any help would be appreciated :)

Nickedemuz, 2019-10-23:
I can see that it tries to find some files for updating but thats not the files in the zip-file:
loading image
----enter to upgrade from sdcard----
    uboot.bin not exist.
    kernel.bin not exist.
    rootfs.bin not exist.
    user.bin not exist.
    app.bin not exist.
----leaved to upgrade from sdcard----

Welcome to root (armv6l-Linux-3.4.43-gk@/dev/ttySGK0/b)
GOKE ARM Linux 3.3  to Copyright (C) 2005 @DaYunLinks

paus56, 2019-10-30:
- what android application does the camera work with?
- Make a firmware backup to the SD card in u-boot:

sf probe 0
sf read 0xC1000000 0x0 0x800000
fatwrite mmc 0:1 0xC1000000 ff.img 0x800000

Nickedemuz, 2019-10-30:
I was mistaken about the brand, it is a Hapsee camera, not an iegeek. 
So the app I’m using is Hapsee. I’m using apple, not android though but I guess it’s the same for android. 
Thanks, I’ll try the fatwrite command. Pretty sure it’s not listed when I run the help command but I’ll give it a try anyway :)

Nickedemuz, 2019-10-30:
Unfortunately I t didn’t recognize the fatwrite command. 
These are the commands I have in I-boot:
[image]
[image]

dotWee, 2019-11-04:
Just received the same camera, I've also used the HapSee application for setup. Have you already made some progress?

Nickedemuz, 2019-11-05:
Sorry, no progress yet. I’ll post here if I have some success :)

daill, 2020-01-22:
Ok, I'm not sure if this is useful but I've extracted the 4 firmware parts (rootfs.bin, user.bin, app.bin and linux.bin). I've managed to have a custom app.bin loaded from sd but I've not yet worked on loading the custom busybox. Furthermore it's possible to run custom sh scripts during the boot process with creating a file called "LinkAfterApi" on the scard, then a script called "after_upgrade.sh" can be run. Sadly there's a reboot directly afterwards. I will open a separate github project soon to have the informations collected.
I own two of these cameras and wanted to have them not communicating to China and run within my home automation w/o using the app.

Nickedemuz, 2020-01-22:
Sounds interesting, it's exactly what I wanted to do with them also.
I'll follow your github project closely and give any input I might have :)

daill, 2020-01-25:
Ok, you can follow this repo https://github.com/daill/h4ps33 if you want to. I'll be working on an automation to have the camera directly connected to the wifi using files from the sdcard ... at least this is the plan. (And the only one for me as weel because I've killed the uart on my second camera by being stupid ;))

Nickedemuz, 2020-01-25:
Following it now :)
And I admit I was “unlucky thinking” when I did one of my cameras as well. Did you know yanking a soldered pin out instead of desoldering it really doesn’t work good at all? I ripped out the close by tracks doing just that 😂😂😂

Skickat från min iPhone

daill, 2020-01-25:
@Nickedemuz feel you ... feel you :)

daill, 2020-02-20:
Ok, I've uploaded a second set of bins and scripts to have a script loaded automatically and disabled some tools which should not run like the rtspd and a tool called "factorytest". Sadly there is no progress to have a free rtsp stream but what I know so far is that they use udp streams. At the moment the bins will start up a tftpd and sshd to have an external connection w/o hustling around with the serial. Not much but baby steps.