ant-thomas/zsgx1hacks #134: Options to adjust/automate night vision

Thorpy, 2019-11-02:
The automatic nightvision function doesn't seem to work, it never switches over from normal to IR.

If I can change the sensitivity of the automatic function or just set it to switch to IR at certain times It'd be nice...or I'd have to keep turning it on and off manually.

Thorpy, 2019-11-02:
The YCC365 Plus app used to control the CloudCam had a setting to "turn the device off" via a schedule, I set a schedule to find the syntax they wanted "Schedule" to be in and added it to the config as:

 <scheduleInfraredLight>                                                                                                     <status>Off</status>                                                                                                              
<schedules />                                                                                                       </scheduleInfraredLight>                                                                                                <nightVision>On</nightVision>                                                                                           <scheduleNightVision>                                                                                                       <status>Off</status>                                                                                                              
<schedules />                                                                                                       </scheduleNightVision>

No more turning night vision on and off :)

closed 2019-11-02 by Thorpy

bleara, 2019-11-15:
I have also been attempting to get a hacked Digoo DG-MYQ to work with IR but without success. On my camera it appears from dmesg that the gpio ports are as follows:-
[    0.170000] [GPIO CFG] IR LED CTL    (53)
[    0.170000] [GPIO CFG] IR CUT1       (53)
[    0.170000] [GPIO CFG] IR CUT2       (44)
[    0.180000] [GPIO CFG] SENSOR Reset  (27)
[    0.180000] [GPIO CFG] PHY Reset     (53)
[    0.190000] [GPIO CFG] PHY Speed Led (55)
[    0.190000] [GPIO CFG] SPI0 EN       (53)
[    0.200000] [GPIO CFG] SPI1 EN       (53)
[    0.200000] [GPIO CFG] USB HOST      (53)
[    0.210000] [GPIO CFG] SD Detect     (32)

but hardinfo.bin seems to inconsistent:-

<BlueLed>35_0x00000000_0_0</BlueLed>
<ReLed>34_0x00000000_0_0</ReLed>
<BoardReset>53_0x00000000_0_0</BoardReset>
<SpeakerCtrl>33_0x00000000_0_0</SpeakerCtrl>
<IrCtrl>36_0x00000000_0_1</IrCtrl>
<IrCut1B>39_0x00000000_0_1</IrCut1B>
<IrCut2B>44_0x00000000_0_1</IrCut2B>

I find that writing a '1' to port 34 turns the blue LED off, ie.
gio -s 34 1
Toggling the state of port 44 gives an audible click so presumably this is the IR cut working.
Port 30 seems to reboot, port 38 and 53 I think are factory resets and indeed when 53 is written to the voice "reset success" is heard, so these are good ports to avoid !

In your last comment are you saying that you have got IR to work? If so are you saying the syntax in the config.xml should be this:-
 
        <scheduleNightVision>
            <status>On Off</status>
            <schedules />
        </scheduleNightVision>

I have tried this to no avail so wonder if this is correct and if there is anything else I have missed?

Thorpy, 2019-11-15:
I've given up trying to turn the IR LED's on, I can control the IRCut and I can turn on "Night Vision" manually in the app which changes the gain of the camera but it does NOT shine any IR light, if there's a car parked near the street light the car's shadow is almost entirely black.

The only time I managed to turn on automatic night vision (without any IR light still) was by selecting the "YCC365 Plus" app option "Device off at" - https://i.imgur.com/qYN7H2C.jpg then taking the settings from "SchedulePowerOff" in the XML settings and putting them in the "ScheduleNightVision" XML settings. I told the app to "turn off" the camera in 1 minute and while watching the rtsp stream I noticed that 1 minute later automatic night vision switched on. The problem with this is it's not persistent and I think a script in the busybox updates the xml values after a certain amount of time because the nightvision never changed again. (except manually changing it like 2 days later via the app) 

YCC365 plus "Automatic night vision" literally just turns on "daymode" and nothing else, I've tried putting the camera in a blacked out box with it on and it still doesn't switch to "nightmode" unless I manually click on it in the app. I thought about testing to see if the LEDs are even real or if I could hijack them to just always be on but the camera was £23 and I'd need to borrow a ladder :( maybe in the future.