ant-thomas/zsgx1hacks #135: Finding the White and IR LED gpio pins to control via gio

Thorpy, 2019-11-03:
I have a sneaky suspicion that the "LED's" on my camera are either for show or they're dead but can't seem to get any lights/switch nightvision on and off via "gio". I'm hesitant to start changing gpio values randomly since some of them "reset" hardware. (pin 53 being BoardReset)

I've tried BlueLed (15), RedLed (15), IrCtrl(12), IrCut1b(44) and 2b(39) both make an audible "CLUNK" but nothing happens to the image.

Gio 12 (IrCtrl) changes from 0 to 1 when using the YCC365 app to turn nightvision off and on but setting it manually via gio -s 12 1 does nothing to the image.

Currently the only ones I've found are:

from hardinfo.bin
<GPIO>                                                                                                                  <BlueLed>15_0x00000000_0_1</BlueLed>                                                                                    <ReLed>15_0x00000000_0_1</ReLed>                                                                                        <BoardReset>53_0x00000000_0_0</BoardReset>                                                                              <SpeakerCtrl>33_0x00000000_0_0</SpeakerCtrl>                                                                            <IrCtrl>12_0x00000000_0_1</IrCtrl>                                                                                      <IrCut1B>44_0x00000000_0_1</IrCut1B>                                                                                    <IrCut2B>39_0x00000000_0_1</IrCut2B>                                                                                    </GPIO> 

from dmesg
[    0.140000] [GPIO]############################# 
[    0.150000] [GPIO] gpio map get from uboot
[    0.150000] [GPIO CFG] gpio   count = 59
[    0.160000] [GPIO CFG] intphy count = 3
[    0.160000] [GPIO CFG] extphy count = 3
[    0.160000] [GPIO CFG] IR LED CTL    (55)
[    0.170000] [GPIO CFG] IR CUT1       (55) 
[    0.170000] [GPIO CFG] IR CUT2       (55)
[    0.180000] [GPIO CFG] SENSOR Reset  (27)
[    0.180000] [GPIO CFG] PHY Reset     (55)
[    0.190000] [GPIO CFG] PHY Speed Led (51)
[    0.190000] [GPIO CFG] SPI0 EN       (55)
[    0.190000] [GPIO CFG] SPI1 EN       (55)
[    0.200000] [GPIO CFG] USB HOST      (55)
[    0.200000] [GPIO CFG] SD Detect     (32)
[    0.210000] [GPIO CFG] SD Power      (55)
[    0.210000] [GPIO CFG] SD1 Detect    (63)
[    0.220000] [GPIO CFG] SD1 Power     (63)
[    0.220000] [GPIO]#############################                                                                      [    0.230000] gpiochip_add: registered GPIOs 0 to 63 on device: gk-gpio0

OptimusGREEN, 2020-12-19:
I'm very interested in this too. I also could only make my night vision purple using gio

laserlance, 2020-12-30:
I have the same settings in /home/hardinfo.bin and I'm able to control the LEDs using gio -s 12 [0 or 1]. I suspect the IRcut filter is in place if the image isn't changing when you turn the LEDs on. To check if the LEDs are lighting, point a cell phone camera at the ip camera and you should see the illumination. 
I think to turn the IRcut filter on, do this: gio -s 44 1 && gio -s 39 0
to turn it off, do this: gio -s 44 0 && gio -s 39 1

dzek69, 2021-02-21:
it seems that the purple image is what we can achieve by manually setting gpios

the monochromatic color is controller by p2pcam and it doesnt seem to care about manual gpios settings