ant-thomas/zsgx1hacks #137: Not able to connect rtsp server

arvindrajput, 2020-01-05:
Device :GD-SC03 
Original firmware: 3.0.6.0801
Tried hacks: Both Read/write and readonly version 
Able to Login both using telnet and SSH
Able to see web gui and tilt the camera using webgui (http://IPAddress:8080/cgi-bin/webui)
nmap on the device ip . All services are running 
Nmap done: 1 IP address (1 host up) scanned in 79.37 seconds
vulcan:Documents ak$ nmap 192.168.1.254
Starting Nmap 7.80 ( https://nmap.org ) at 2020-01-05 10:58 CST
Nmap scan report for 192.168.1.254
Host is up (0.013s latency).
Not shown: 990 closed ports
PORT     STATE SERVICE
21/tcp   open  ftp
22/tcp   open  ssh
23/tcp   open  telnet
80/tcp   open  http
554/tcp  open  rtsp
843/tcp  open  unknown
5050/tcp open  mmcc
7103/tcp open  unknown
8001/tcp open  vcom-tunnel
8080/tcp open  http-proxy

Issue unable to connect to rtsp used following links

rtsp://192.168.1.254/
rtsp://admin:admin@192.168.1.254/
rtsp://192.168.1.254:554
rtsp://admin:admin@192.168.1.254:554
rtsp://192.168.1.254:554/0/av0
rtsp://admin:admin@192.168.1.254:554/0/av0

Nothing works. 

Thanks
Arvind

naturaldisasta, 2020-01-10:
I know this is simple but have you tried:

rtsp://admin:@192.168.1.254/0/av0

(admin without password)

video2go, 2021-03-10:
I tried almost anything from Google, but I can not get RTSP stream on my zsgx1. I run readonly hack.

grega