ant-thomas/zsgx1hacks #139: Hello

Anon0ne, 2020-01-14:
I have a INQMEGA PTZ381 that I have started to work on. I got the cloud disabled, and managed to get the ONVIF working. I had made a few changes this morning and backed up everything before I started.

Now, after installing the latest busybox and starting ssh/ftp and such a few times the device will no longer connect to the net via WiFi; I can only get it to connect via Ethernet. Any ideas what I may have done that would make it not connect via WiFi? I am going to hookup to lan and investigate further, but it's irritating.

Anon0ne, 2020-01-14:
It is starting the AP now, that was there for the original setup of the device.

Anon0ne, 2020-01-14:
Okay, I figured it out. It seemed as if the p2pcam instance has taken it's own liberty to remove the config.xml and devParam.dat files from the home folder! I had them backed-up and made some symlinks to ensure it's always good. seemed to connect to wifi again; let's see if the debug_cmd will work again and start our servers.

KalastrianP, 2023-03-10:
> 

Know its been quite a while, but I just got a few of these old ipcams, and NONE of them connects to ycc365, no matter which mode I try.

Previous owner was a real tech novice, and I think somehow the firmware got corrupted.

I can connect to it by ONVIF and RTSP, by cable or WIFI, but I cant get it to connect to the YCC app or to record to the SDCARD.

Also, before using the readonly hack, the ipcam had no ssh, telnet, or even http. Thankfully it does now, with the hack, but the rest of the problems remain.

Would you be so kind to share your firmware/config? I'd love to get these things working.

Thank you

KalastrianP, 2023-03-10:
> Okay, I figured it out. It seemed as if the p2pcam instance has taken it's own liberty to remove the config.xml and devParam.dat files from the home folder! I had them backed-up and made some symlinks to ensure it's always good. seemed to connect to wifi again; let's see if the debug_cmd will work again and start our servers.
> 
> NOTE: THERE IS NEWER VERSION OF FIRMWARE AND SDK THAT WILL BREAK WITH THESE HACKS FOR GC2053, GK7102. I have a bCkup I will share for who needs once I edit personal details.
> 
> YOU REALLY NEED TO WARN PEOPLE! MAKE A COPY OF YOUR HOME DIRECTORY, ESPECIALLY BACKUP devParam.dat once your WiFi is setup.

> 

Know its been quite a while, but I just got a few of these old ipcams, and NONE of them connects to ycc365, no matter which mode I try.

Previous owner was a real tech novice, and I think somehow the firmware got corrupted.

I can connect to it by ONVIF and RTSP, by cable or WIFI, but I cant get it to connect to the YCC app or to record to the SDCARD.

Also, before using the readonly hack, the ipcam had no ssh, telnet, or even http. Thankfully it does now, with the hack, but the rest of the problems remain.

Would you be so kind to share your firmware/config? I'd love to get these things working.

Thank you