ant-thomas/zsgx1hacks #140: DG-W01f Bricked. Without wanting to connect after hack. Could anyone help?

vinnywebmaster, 2020-02-06:
My camera had the same problems as this post:

[Device is not booting (bootloop) #44](https://github.com/ant-thomas/zsgx1hacks/issues/44)

he device did not respond after the hack, it looked dead. However, I followed the instructions of others in the post and was asked to execute the debug_cmd that @ant-thomas passed, and just below an attempt to execute start_sh.
Okay, I put the files on a memory card along with firmware.bin, ruishi-38-hwcfg.ini and OVESEA-VOICE.tgz (available on the Digoo website to change the password). After that, the camera returned to give life signal (the RJ45 cable connector light flashes green and then remains stable, as if maintaining any connection). However, the camera does not connect at all, and the voice is also gone. No more sound, just popping sounds when it starts. When I put my hand closing the lens, the infrared lights come on, so I know that the camera is probably working but it doesn't connect or emit sound and having a device like this is the same as having a useless piece of iron.

[image]

I also tried to pass the IP tracker on my network and no additional ip of the camera appears.

Could someone help me re-establish the camera's original state or install the hack so that it leaves the camera running?

(Sorry for my English, thanks to everyone).

OptimusGREEN, 2020-12-20:
I would also like to know how to flash a firmware

vinnywebmaster, 2020-12-20:
> I would also like to know how to flash a firmware

With this comment you are stating that you know how to record the flash on this device. Rather, you are saying that I don't know how to record a flash and you are guaranteeing that it will work on this device. Since you are so smart, the whole community is wondering how you do it. Teach us!

OptimusGREEN, 2020-12-20:
> With this comment you are stating that you know how to record the flash on this device. Rather, you are saying that I don't know how to record a flash and you are guaranteeing that it will work on [...]

??? No, It is clearly stating that I don't know how to flash the device. strange reply dude.

krcenov, 2021-04-13:
Maybe there is a serial connector on the board.
Try connecting to that.