ant-thomas/zsgx1hacks #143: How to turn off led’s

kkwwm, 2020-02-25:
Hi,
i have read only hack on my outdoor IP camera. My question is how to disable led light (floodlight, the normal, white leds).
Thanks a lot!

dzek69, 2021-02-21:
you can turn off the white led, but it won't make camera automatically use the IR and take off ir cut when needed. Also the picture will be purple, because the binary p2pcam software won't make the image monochromatic and nobody yet found a way to communicate with it (it has so many open ports there sure is a way ;))

check your hardinfo.bin, you will have `WhiteLight` XML element, the first number, before the underscore is your white LED gpio number

turn it off with `gio -s WhiteLight_Number 0`

if there is dark (nobody found a way to read the light sensor yet) you may manually turn on IR diode and remove the IR light filter (you want to have it on during the day, because the colors will be washed out if IR filter is on durinig the daylight)

so
`IrCtrl` is the IR diode - turn it on with `gio -s IrCtrl_Number 1`

so take off the IR filter:
`gio -s IrCut1B 0 && gio -s IrCut2B 1`
then apply 
`gio -s IrCut2B 0`
to release the "pressure" on the mechanism

and when there is daylight turn off IR diode: `gio -s IrCtrl_Number 0`
take on IR filter:
`gio -s IrCut1B 1 && gio -s IrCut2B 0`
and release:
`gio -s IrCut1B 0`

I hope after a year it is still useful to you :) i've kept my camera in a shelf for 2 years before i finally started to hack it so there is a chance

dzek69, 2021-02-21:
one more thing - this camera turns the white led back on randomly when there is darkness