ant-thomas/zsgx1hacks #144: Sonoff GK-200MP2-B

onedayfishsale, 2020-03-15:
The Sonoff GK-200MP2-B is a GK7102S based camera that uses the eWeLink app for configuration and control. I tried the readonly and read/write hacks with no success. Unfortunately I don't think there's been a firmware update from Sonoff since the camera was released either.

After WiFi setup (or plugging in to Ethernet) a telnet server is running on 23/tcp but I can't figure out the password (not root/cxlinux). The GK-200MP2-B board has a well labeled programming header and I successfully connected at 115200 baud (leave 3V3 disconnected if your FTDI supplies it). Unfortunately, it still wants a username and password at the serial console.

I'm just dropping this here for visibility and intend to update if I make progress. If there's a better place to chat about the GK-200MP2-B let me know.

https://www.itead.cc/sonoff-gk-200mp2-b-wi-fi-wireless-ip-security-camera.html
https://sonoff.tech/ewelink

EpicLPer, 2020-04-03:
Hey, I created a repo with a dumped firmware and further infos if you’re interested! I’m not a programmer myself so I hope some genius minds can hack it further from there :)

onedayfishsale, 2020-04-03:
@EpicLPer nice work! I'll definitely take a look at your repo (https://github.com/EpicLPer/Sonoff_GK-200MP2-B_Dump) and see if we can get zsgx1hacks working on it. Thanks for the heads-up!