ant-thomas/zsgx1hacks #145: ONVIF device_manager not found

flsabourin, 2020-03-30:
Hello
I am running with 3.4.2.0422 firmware

My hwcfg.ini is showing:
`support_onvif = 1`

However, I can't manage to discover the camera capabilities with an onvif client
`http://robotcam.lan:80/onvif/device_service`
is not giving any reply, nor for any other port.

Did someone found the URI for the device_service?

PS: all other features are working correctly

Thanks

EpicLPer, 2020-04-03:
I kinda think the project is dead sadly

lildigiman, 2020-04-16:
It appears cameras that are able to work with this firmware hack don't have full ONVIF support at all. I installed this hack on my camera and I only had limited success discovering the rtsp stream (using ONVIF Device Manager v2.2.250) and it was incapable of doing much else.

indian5, 2021-12-05:
I see it's a old post, but maybe helps somebody. What wsdl do you use? 
Maybe try this as soap request body:
```
<soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope" xmlns:wsdl="http://www.onvif.org/ver10/device/wsdl">
   <soap:Header/>
   <soap:Body>
      <wsdl:GetServiceCapabilities/>
   </soap:Body>
</soap:Envelope>
```