ant-thomas/zsgx1hacks #147: SD Card Recording

rsauve, 2020-04-30:
Has anyone been able to get these cameras to record to sd card while blocking the cloud service?

Blocked cloud services are listed in /media/hack/host.new
My camera uses:
127.0.0.1 esd.icloseli.com

Unblock this entry and the camera will record automatically to the sd card, block it and the camera stops recording to the sd card.

I have had a look into start.sh, config.xml, config.cfg, pretty much all the files in the /home directory.
Nothing that seems to work. Perhaps this function is tied up in the binarys?

Camera info:
Big+1080p
https://www.amazon.com.au/Wireless-SmartSens-Security-Surveillance-Speaker/dp/B07MNZZNCT
Hardware model
GOKE 7102C BOARD
Firware Version: 3.4.2.0923
gc2053
chip=gk7102s

MarcBoettinger, 2020-12-11:
Interesting point. I haven't applied the hack but my camera does record to the sd card (without blocking anything) but will only playback per app one single time. Thereafter no access to the sd card. A reboot per telnet does not help nor an reboot per app. I have to power off / on.

laserlance, 2020-12-30:
Yes, essentially the camera gets it's settings from the "cloud" server, so one needs to perhaps make their own server and point to it in /home/cloud_oversea.ini where the esd.icloseli.com entry is. On your server, emulate the responses from the real server for each endpoint. The only setting that appears to need to be correct is the time. 

Here are the endpoints that I've found, in order of execution:

1. /lecam/openapiSig/device/deviceListNew
2. /lecam/sigV2/support/getUtcTime 
3. /lecam/sigV2/profile/getCurrentSetting
4. /lecam/sigV2/profile/saveSettingAttrByPaths
5. /lookup/sigV2/assignRelayIp
6. /lecam/sigV2/andmu/setLocalPlayInfo

The responses need to be JSON, non-gzipped. "getCurrentSetting" is XML wrapped in JSON and holds the settings that are set when using the application.

Once you return the responses after each of these calls, the camera should record to the SD card. Note that if the responses are not correct the camera will enter a startup-shutdown loop until you deliver the correct data via the endpoint. Also, I don't know if it matters, but I left the default values for the "assignRelayIp" call in, which streams the video to the Chinese server (where you can use the YCC365 app to view it when you're not at home).

WebWire-NL, 2021-04-05:
Hi,
I have this same problem, i'm trying to make an alternative option to upload to the cloud... uploading to google drive. But whenever i apply the hack the app tells me there's no memory card inserted... 

I inserted the card and started recording some events, then i powered off the camera, mounted the sdcard and applied the patch... it doesn't matter if enable HACK_CLOUD or disable it, the sd card management tells the sdcard is unavailable. Do you have any idea? I really want to make a cloud alternative.... I'm so close :)

csurf, 2021-06-07:
@WebWire-NL 
I posted new issue here, #161, before seeing this issue.

I'm running into the same problem. I believe that perhaps the p2pcam app (in newer firmwares?) will purposefully disable the SD card if it finds any 'hack scripts / files' stored on it, but I'm not sure how to confirm this. 

I did notice was that there's a message in the '/tmp/closelicamera.log' file that mentions sending the SD card info up to the cloud service, something to the effect of calling 'getSDcardInfo' (not the exact name of the routine, I didn't save the message, but you get the gist). 

One thought that came to mind is that perhaps the p2pcam app sends a directory listing of the SD card contents up to the cloud, and the cloud server decides whether or not to enable the card access based on the card's contents. Another idea might be that p2pcam simply decides to disable the card if it finds a 'debug_cmd.sh' script is on it. However, these are both just guesses on my part at this point.

It would be great if someone here who has reverse engineering skills could take a look at the p2pcam app and see if they can find anything related to the SD card & its contents. 

In my case, I simply wanted to use the readonly hack / 'debug_cmd.sh' script in order to enable SSH and FTP access for better management & to be able to retrieve the stored motion videos off of the SD card without having to use the app. However, this might be a deal breaker if there's no way to load the debug script & still enable motion recording access to the card.

WebWire-NL, 2021-06-07:
I already fixed it.... The daemon unloads the sdcard.... So if you mount
the sdcard to a different location and start Ur scripts from there then
there's no issue anymore.... I made a small extra partition on the sdcard
and mounted this second partition on a different location and started my
scripts from there.
..

WebWire-NL, 2021-06-07:
Forgot to mention that I made a symlink to the device also... Don't
remember exactly.... But let's say sdcard is /dev/mmc I made a symlink and
called it /dev/mysdcard.... And then mounted that one....

csurf, 2021-06-08:
> I already fixed it.... The daemon unloads the sdcard.... So if you mount the sdcard to a different location and start Ur scripts from there then there's no issue anymore.... I made a small extra [...]

I tried following your advice, creating two partitions, the first for data, mmcblk0p1, and the second for the readonly hack files, mmcblk0p2. I was able to mount the 2nd partion from the debug_cmd.sh script, but it still didn't work. The SD card shows up as unavailable, and the mmcblk0p1 partition doesn't get mounted. 

Any ideas?

Also, what did you mean by having to create a symlink for the device file? why would this be necessary?

WebWire-NL, 2021-06-08:
I'll try to lookup the relevant code soon....

csurf, 2021-06-08:
@WebWire-NL 

I ended up doing things a bit differently. 

My solution was to setup the debug script to run in a background shell, and have it sleep for a while before running all of the setup commands. 

I also avoided trying to mount the SD card myself, and instead relied on the mount points that p2pcam creates, `/tmp/mnt/mmc01/0/` and `/tmp/mnt/mmc01/1/` for partitions 1 and 2 respectively. 

So, in my case, since I stored my hack scripts on the 2nd partition, I adjusted all of the paths within `debug_cmd.sh` to use the `/tmp/mnt/mmc01/1/` path as the root directory for each hack file/command.

Within `debug_cmd.sh`, I did something like this:
```
#!/bin/sh
(
    sleep 60 &&
    (
      # all of my other 'debug_cmd.sh' command logic goes here
    )
) &
```
I'll have to play with the sleep time value in order to see what's the minimum amount of time that can be used safely in order to allow the p2pcam process to properly initialize the mount points. In my case, 60 seconds seemed to work fine as a first guess.

I also need to test to see if this 'async delay' strategy will work when using a single SD card partition to both store the hack files & the local AV recordings.

WebWire-NL, 2021-06-08:
Thats allmost the same what i did :) bit i checked if the daemon was
already running or not.... Check my script.... It alsof runs a background
script

csurf, 2021-06-09:
@WebWire-NL 
Did you post a script? I didn't see anything. no download links. Can you try posting again?

WebWire-NL, 2021-06-09:
I attached it tot one of my mails earlier....

rsauve, 2021-06-17:
I do not believe that this hack will be complete until the camera can run independently from the associated cloud service. At this point in time if the cloud service is block from communicating with the camera, recording to the local SD card is disabled.
SD card recording files are stored as: /mnt/yyyymmss-hhmmssM.mp4

This is from mt /etc/ hosts file:
# This is the one! Un-comment out to block
127.0.0.1 esd.icloseli.com

Entry from my /tmp/closelicamera.log file when the cloud service in disabled:
-----------------------------------------------------------------------------------------------------------------------------------
2021-06-15 19:39:21.886729 [299] [HttpHandle.cpp(142) handle]Error: 7 - Couldn't connect to server -- https://esd.icloseli.com./lecam/openapiSig/device/deviceListNew
2021-06-15 19:39:22.045873 [283] [RtspBufferManager.cpp(811) CreateNode]Trace: [0:0] idle=111,buffer=510,create=621,destroy=0, relay
2021-06-15 19:39:22.047303 [283] [RtspBufferManager.cpp(811) CreateNode]Trace: [0:0] idle=175,buffer=322,create=497,destroy=0, p2plive
2021-06-15 19:39:22.929920 [283] [DeviceStream.cpp(279) statistics]Trace: v[0,0](212.22/226.73)Kbps 11.84fps, PTS(212.22/226.73)Kbps 11.84fps
2021-06-15 19:39:22.931825 [283] [RealStream.cpp(3915) SetExternalStreamData]Trace: 1df5cc0[0:0][1920*1080]: Get I cpacket external, size: 39764, starttime: 26152463, num: 307106, frames: 36, at 1623749962931, fisheye:0
2021-06-15 19:39:23.064152 [299] [OpenCore.cpp(425) IsRegistered]Trace: username(<subscriber's email address>)
2021-06-15 19:39:23.069647 [299] [LECAMImpl.cpp(460) FormatURLString]Info: <<<<<< size: 119 <<<<<<<<<<<< deviceId=xxxxS_7ca7b0888b83&token=NULL&apiVersion=1.0&accessKey=6f425be2-e27&signature=7ea9baec76cfd9ca81ea1be0c3c25390
2021-06-15 19:39:23.072794 [299] [HttpHandle.cpp(118) handle]Info: CURL_SSLVERSION_TLSv1_2 -- /p2pcam/ca-bundle-add-closeli.crt
-----------------------------------------------------------------------------------------------------------------------------------
The above entry repeats.

This is what the /tmp/closelicamera.log file looks like when the cloud service is enable:
-----------------------------------------------------------------------------------------------------------------------------------
2021-06-15 19:49:43.600637 [258] [MemoryPool.cpp(77) alloc_cells]Trace: this(0x2099170) cell_size(512) pool_bytes(263680) all_bytes(2196992)
2021-06-15 19:49:43.603573 [258] [MemoryPool.cpp(77) alloc_cells]Trace: this(0x2099170) cell_size(512) pool_bytes(264192) all_bytes(2197504)
2021-06-15 19:49:44.683555 [282] [RtspBufferManager.cpp(811) CreateNode]Trace: [0:0] idle=123,buffer=498,create=621,destroy=0, relay
2021-06-15 19:49:44.763435 [282] [RtspBufferManager.cpp(811) CreateNode]Trace: [0:0] idle=178,buffer=319,create=497,destroy=0, p2plive
2021-06-15 19:49:45.256877 [282] [RtspBufferManager.cpp(262) HandleWriteBufferSmartAVC]Trace: 35856112: streamstat: total size:9944(Bytes), timespan:10000(ms), count:536.
2021-06-15 19:49:45.258161 [282] [RtspBufferManager.cpp(262) HandleWriteBufferSmartAVC]Trace: 35867280: streamstat: total size:6528(Bytes), timespan:8000(ms), count:352.
2021-06-15 19:49:45.660430 [282] [DeviceStream.cpp(279) statistics]Trace: v[0,0](147.86/184.07)Kbps 11.84fps, PTS(147.86/184.06)Kbps 11.84fps
2021-06-15 19:49:45.662338 [282] [RealStream.cpp(3915) SetExternalStreamData]Trace: 2231978[0:0][1920*1080]: Get I cpacket external, size: 39397, starttime: 421733, num: 4545, frames: 36, at 1623750585662, fisheye:0
2021-06-15 19:49:46.122210 [278] [APDirectConnect.cpp(240) udpListenProcess]Trace: select timeout
-----------------------------------------------------------------------------------------------------------------------------------
The above entry repeats.

Ideally this device should be able to run independently. Power up the camera and it records automactically to SD card. It should be able to do this regardless if it communicates with the cloud service or even has a wifi connection.

To date I have not been able to hunt down the code that controls the cloud service verification and SD card recording enable/disable function.

WebWire-NL, 2021-06-19:
> 0,0 <http://212.22/226.73>Kbps 11.84fps, PTS(212.22/226.73)Kbps 11.84fps
> 0,0 <http://147.86/184.07>Kbps 11.84fps, PTS(147.86/184.06)Kbps 11.84fps

Feel free to write a new camera daemon.... It is possible without
cloud..... You have tot write a web interface.... Which just plays your
files.... This is just a new version of an old camera.... Its not hard to
send the mp4 files to cloud.... I lost interest..... BC of people
complaining and not rewarding in open source projects ;)

csurf, 2021-06-25:
I am very interested in the possibility of coding a new camera daemon to replace p2pcam, but no one seems to want to share the SDK files with me. It almost seems as if some users found the SDK files, initially decided to share them publicly, but now are trying to monetize the files.

If someone has the GOKE SDK files, please send me a msg if you're willing to share them.

bitfarmer, 2021-06-26:
I believe I have them somewhere and I will certainly make them available to you! I might even be interested in becoming involved with your project.

phicha20224, 2021-10-02:
need to format sdcard from app, wont work format from windows.