ant-thomas/zsgx1hacks #154: EYEPLUS, EYEPLUS_DEV CLOUDCAM not accesible through ftp

jpslootmans, 2020-12-11:
chip: Goke GK7102
Firmware: 3.4.2.0923
camera connected via ethernet
used sdcard with "readonlyhack-v0.1"
no ftp connection possible
via telnet checked if files are transferred to /media: no, empty:
[image]
Anybody ideas how to proceed?

fabriciogmatos, 2020-12-11:
hello, my firmware is still the original but i believe you can see all the commands available on the device using the `busybox` **command
**
There are two commands there that I use:

```
ftpput
fptget

```
ftpput - to send files to an ftp server on my local network.
fptget - to download files from the ftp server to my camera.

I don't know if I'm helping anything else, I'm available.

[image]

jpslootmans, 2020-12-11:
@fabriciogmatos 
thank you, that seems like a way out!
what would the syntax be if I want to transfer the "devParam.dat" file to my PC?
Thanks in advance for your time,
JP

fabriciogmatos, 2020-12-11:
Hello my friend, use the syntax below.

use your ftp server username, password and IP.

` ftpput -v -u ftp_user -p ftp_pass ftp_server_IP  devParam.dat`

I am available to help in any way I can.

jpslootmans, 2020-12-11:
Hello Fabricio,
the ftp issue is solved. With your help I managed to connect the camera and upload a new devParam.dat.
But, .... I hoped to solve the rtsp connection problem by uploading a devParam.dat with an empty Onvif password.
I only succeeded to shut myself out. No more telnet and according to Zenmap the host is down. I think the camara is beyond repair.

maartenjd, 2020-12-12:
Don't panic. Go for the serial hack. But you'll probably be needing some soldering skills.

jpslootmans, 2020-12-12:
@maartenjd 
can you explain the "serial hack" or send a link?

jpslootmans, 2020-12-12:
@fabriciogmatos 
I searched the internet for my camera make and type and came across many useful sites, but as you noticed I managed to brick my camera in the end. If I was not able to view the videostream in my domotic system, I was going to throw it away anyway. Not one hair on my head believes I will expose my personal data to a Chinese website or pay for a service I will probably never need.

maartenjd, 2020-12-12:
https://github.com/jpslootmans https://github.com/ant-thomas/zsgx1hacks/issues/132

jpslootmans, 2020-12-13:
thank you maarten.
The link provides information on how to access the serial port, but I can not find a description how to flash the firmware.
If anybody can help, greatly appreciated.
JP

maartenjd, 2020-12-14:
Hello [JP](https://github.com/jpslootmans),

I bricked my camera and used the following procedure.

Do you have an image of the original firmware? Then you can use that fw-image. What size is it? Did you previously determine the flash layout?

Once you have a serial connection you have to determine the flash layout. You will see it during the boot process and it will look something like:

> Creating 5 MTD partitions on "gk_flash":
> [    0.640000] 0x000000000000-0x000000050000 : "U"
> [    0.650000] 0x000000050000-0x0000001f0000 : "K"
> [    0.660000] 0x0000001f0000-0x000000310000 : "R"
> [    0.670000] 0x000000310000-0x000000590000 : "A"
> [    0.680000] 0x000000590000-0x000000800000 : "H"

Anyhow, I downloaded [HIP291G_Mod_ENG_04102019.ZIP](https://github.com/ant-thomas/zsgx1hacks/files/3694126/HIP291G_Mod_ENG_04102019.ZIP) and unpacked it because my firmware was corrupted. You should see a firmware image of 8.4 Mb (Hip_Mod_fw.img). Rename it to fw.img for ease of things.

I used binwalk to find out what the flash layout of this fw is. It appears to be different from my original fw but in the end it did not matter. I cut up the fw into pieces using the sizes I determined with binwalk:

> dd if=fw.img of=fw0.img bs=1024 count=320
> dd if=fw.img of=fw1.img bs=1024 count=1664 skip=320
> dd if=fw.img of=fw2.img bs=1024 count=1856 skip=1984
> dd if=fw.img of=fw3.img bs=1024 count=320 skip=3840
> dd if=fw.img of=fw4.img bs=1024 count=4032 skip=4160

I did that to avoid flashing mtd0 (where u-boot resides). Then I (re-)merged the mtd1-4 partitions:

> cat fw1.img fw2.img fw3.img fw4.img > fw1-4.img

I did that to avoid flashing mtd0 (where u-boot resides). I could have made it easier in the previous step by doing 

> dd if=fw.img of=**fw1-4**.img bs=1024 count=7872 skip=320

right away but I wanted to be able to inspect the different partitions to see if I could hack them. But that is a different story.

I used Kermit as the terminal program and connect it to the serial port. Start it from the directory where the FW-file is  and switch to terminal mode (C). Power-up the camera and you should see the boot process. Interrupt the boot process (it says "press enter to interrupt but sometimes you have only one second to do that) and you will enter the U-boot environment. From there you can use flash commands.

In u-boot (you can type "help" to see the possible commands and their functions). Start with loadb to set it in file transfer mode:

> loadb

The fw-file now sits in RAM at address 0x1000000. Switch to kermit command mode and type:

> send **fw1-4**.img

Back to terminal mode:

> sf probe
> sf erase 0x50000 0x7b0000
> sf write 0xc1000000 0x50000 0x7b0000

To write, starting at 0x50000 and for 0x7b0000 bytes the contents in RAM to flash. The fw should be flashed now. Reboot/restart the camera.

jpslootmans, 2020-12-18:
@Maarten,

wow, that's quite something. Thanks for the procedure.
I have no informating about the flash whatsoever, so it will be trial and error.
The first step will be to find Rx and Tx for the serial connection and connect my Raspi.
I'll start next week, when there's more time.
I found out that the camera is not completely dead, because when I put an empty SD card in, it writes an INFO.txt with the contents: "eye.conf already existed !"

Regards,
JP

maartenjd, 2020-12-18:
Hello [JP](https://github.com/jpslootmans),

Careful with the serial pins as voltages may be different (5V or 3.3V) and you could damage your camera or you Raspi. Better use a [USB2serial-cable](https://nl.aliexpress.com/item/32775876897.html)  or an [FTDI-adapter](https://nl.aliexpress.com/item/32634246159.html). 

Send me a picture of your camera's motherboard.

Maarten

jpslootmans, 2020-12-27:
Hello Maarten,

finaly found some time to put the camera back on the table.
Here are the pictures:
[image]
[image]

maartenjd, 2020-12-28:
Hello [JP](https://github.com/jpslootmans),
It looks as if J3 on the second picture is a com-port (RX/TX) to which you could attach the TX/RX-connectors of a (usb2)serial cable. You only need to find a proper GND but that should be no problem. By the way: you have a GK7102S, not a GK7102.
Maarten