ant-thomas/zsgx1hacks #155: EYEPLUS camera configuration

pouilld, 2021-01-19:
Hi,
After having installed this cloud camera using the YCC365 plus android utility, I have installed the readonlyhack-v0.1 on the SDcard of this camera. I can actually 
 - ssh into the camera
 - prevent camera cloud connection and avoid cloud connection to the camera from outside by using the host.new file in the hack folder
 - 
I noticed that the camera Wifi parameters (SSID/pdwd) can be specified in a cls.conf file.
So I would like to know 
- what is the place to store this cls.conf file in the SD card (it seems not to work when placed at the root of the SD card.)?
- is it possible to configure an "out of the box camera" using the hack to define camera parameters without the need to "activate" by creating an account on a cloud server and setup the camera with a specific smartphone application?
Thanks for your answer.
D. Pouilloux

mason-xx, 2021-04-20:
got the same one ... are you able to get a rtsp stream out of it?

zu3st-de, 2022-05-11:
I've an escam pt208 which seems to be an eyeplus. Any Updates?

tymoyato, 2022-07-02:
> mason-xx

How do you ssh please