ant-thomas/zsgx1hacks #156: GK7102S - Firmware CH341A !

XSF7, 2021-02-11:
Guys i need help ,i installed the "HIP291G_Mod" firmware on my "IL-HIP291G-2M" via Sd card.
the camera failed to reboot, then i used the CP2102 to check the boot sequence (uboot) and all i get is "UGKMC" 
also i used CH341A programmer to read the flash chip "Winbond 25064jvs" the output was "FF FF FF" that means empty
can someone please share with me the firmware as a bin file from the flash ship ?

svastarija, 2021-02-19:
I am also looking for Hip_Mod_20191103_ENG_7102S flash file.

XSF7, 2021-02-19:
I found a flash for GK7102S ,it works
use an app like ILUUK ,YSEE to pair the camera
[GK7102S_Mod+Camera.zip](https://github.com/ant-thomas/zsgx1hacks/files/6009463/GK7102S_Mod%2BCamera.zip)

closed 2021-02-19 by XSF7