ant-thomas/zsgx1hacks #159: Does not connect to WIFI

krcenov, 2021-04-10:
The camera does not want to connect to the wifi, but if i do
```
ip link set wlan0 up
iwconfig wlan0 essid "K!DA6A 2.4GHZ"
wpa_supplicant -c /tmp/mnt/mmc01/0/wpa_supplicant.conf -i wlan0
```
it connects and works like a charm

**wpa_supplicant.conf** - contains the SSID of the network and the psk. I have created it in the memory card
my firmware is 3.4 read only
```
network={
        ssid="home"
        scan_ssid=1
        key_mgmt=WPA-PSK
        psk="very secret passphrase"
}
```
Is there a way to automatically run this command after camera initialisation?

sugakenn, 2021-05-20:
Hello! krcenov

I encounted same probrem.
How about this code.
It works my enviroment.

1.make .sh file and put it into the dir that is belong to debug_cmd.sh.

2.edit debug_cmd.sh last row.
(sleep 30 && changenetwork.sh) &

#!/bin/sh
#
# changenetwork.sh

# change wifi to ownnet-work

# add wpa_supplicant.conf
cat << EOS >> /tmp/wpa_supplicant.conf
update_config=1
network={
        ssid="..."
        psk="..."
}

EOS

# kill dhcpd
pkill udhcpd

# kill hostapd
pkill hostapd

# down wlan0
ip link set wlan0 down

# not found dhclient, so change ip address manually
ifconfig wlan0 192.168.1.155

# wlan0 up
ip link set wlan0 up

sleep 5

# wpa_supplicant start
wpa_supplicant -Dwext -iwlan0 -c/tmp/wpa_supplicant.conf -B

krcenov, 2021-05-20:
Will try!
Thank you :)

hondcxrider, 2026-01-14:
I tried this and got:
ioctl[SIOCSIWAP]: Operation not permitted
can you help?