ant-thomas/zsgx1hacks #161: SDCard unavailable in app when readonly hack is applied

csurf, 2021-06-07:
I've got a 1080p outdoor ptz camera running firmware 5.5.4.0728. The readonly SD card hack using the 'debug_cmd.sh' script seems to work fine; I'm able to login via SSH & probe around the system, and everything looks good. I can even disable the hosts hack & allow the camera to communicate with the 'closeli' cloud servers in order to use the app to configure the camera.

When using the readonly hack off of the SD card, the cloud app shows the SD card as being unavailable, even though it's clearly mounted & available. Obviously, the fact that the SD card is being reported as being unavailable also disables the ability to record motion events to the SD Card.

Has anyone else run into this issue? Could it be that this newer version of p2pcam is checking to see if the hack has been applied to the SD Card & purposefully disabling the card if it finds anything there? I looked at the built-in start-up scripts, and I didn't see anything that looked like it could cause issues or change the way that the SD Card is mounted & presented to the system.

csurf, 2021-06-07:
FYI, also related to #147...

albenex, 2021-06-09:
same too here.
i have 2 different cameras.
both support ZSGX1 HACK, one has app "YCC365" and it's fine on playback. The other one has app "AISEE" and says SD card is corrupted while trying to play recorded events even if it is correctly recognized (capacity, space left etc..)

albenex, 2021-11-11:
> I've got a 1080p outdoor ptz camera running firmware 5.5.4.0728. The readonly SD card hack using the 'debug_cmd.sh' script seems to work fine; I'm able to login via SSH & probe around the system, and everything looks good. I can even disable the hosts file hack & allow the camera to communicate with the 'closeli' cloud servers in order to use the app to configure the camera.
> 
> However, when using the readonly hack off of the SD card, the cloud app shows the SD card as being unavailable, even though it's clearly mounted & available. Obviously, the fact that the SD card is being reported as being unavailable also disables the ability to record motion events to the SD Card.
> 
> Has anyone else run into this issue? Could it be that this newer version of p2pcam is checking to see if the hack has been applied to the SD Card & purposefully disabling the card if it finds anything there? I looked at the built-in start-up scripts, and I didn't see anything that looked like it could cause issues or change the way that the SD Card is mounted & presented to the system.

ok after digging with camera for other reason i found that AISEE app was very similar to TUYA Smart Life.. so I paired camera with Tuya and now both readonly and SD recordings works.