ant-thomas/zsgx1hacks #162: How to change default RTSP port

albenex, 2021-06-07:
Hi,
i've 2 of these cameras and i would control them remotely.
unfortunately my router can't map internal/external ports with different port number so I need to change default RTSP port (554) to a different one.
is this possible?

otherwise a workaround could be to transcode stream with a local raspberry and send it to a different socket.
however in this case i'd lost streams if raspberry freezes.

thank you in advance.

viniciusvillas, 2021-07-02:
Hello albenex! You can try the port 8001. In my camera, it's a clone of port 554.
```
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 0.0.0.0:7101            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:7103            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:8001            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:3201            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:554             0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:6670            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:23              0.0.0.0:*               LISTEN
```
Another option is a little more complicated: you can use the dropbearmulti binary available in this github repository to connect to a SSH server in your raspberry PI. Using the dropbear client (dbclient), open a SSH tunnel mapping the remote port on you raspberry, say, 64554, to the local 554 port. To avoid being asked for credentials, you export the environment variable `DROPBEAR_PASSWORD` and dbclient will use it automagically!

I do something very similar with the following code inside `debug_cmd.sh`. It has been working like a charm!
```
# Custom Environment Variables.
export DROPBEAR_PASSWORD=<password>

# Enable SSH Tunnel.
while true
do
  sleep 60 # Wait 60 secs and repeat every minute.
  # Check whether dbclient is running.
  if ! ps | grep "dropbearmulti\sdbclient" >/dev/null
  then
    /mnt/mmc01/0/bin/dropbearmulti dbclient -N -f -y -g -R 8001:localhost:8001 <user>@<server_ip_or_dns>/<ssh_port>
  fi
done &
```
I actually use it to get past a NAT444 that my provider has deployed on 4G networks. My first option was using a VPN (like WireGuard), but I couldn't find a client already compiled for this architecture (ARMv6), unfortunately.

albenex, 2021-07-03:
> Hello albenex! You can try the port 8001. In my camera, it's a clone of port 554.
> 
> ```
> Active Internet connections (servers and established)
> Proto Recv-Q Send-Q Local Address           Foreign Address         State
> tcp        0      0 0.0.0.0:7101            0.0.0.0:*               LISTEN
> tcp        0      0 0.0.0.0:7103            0.0.0.0:*               LISTEN
> tcp        0      0 0.0.0.0:8001            0.0.0.0:*               LISTEN
> tcp        0      0 0.0.0.0:3201            0.0.0.0:*               LISTEN
> tcp        0      0 0.0.0.0:554             0.0.0.0:*               LISTEN
> tcp        0      0 0.0.0.0:6670            0.0.0.0:*               LISTEN
> tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN
> tcp        0      0 0.0.0.0:23              0.0.0.0:*               LISTEN
> ```
> 
> Another option is a little more complicated: you can use the dropbearmulti binary available in this github repository to connect to a SSH server in your raspberry PI. Using the dropbear client (dbclient), open a SSH tunnel mapping the remote port on you raspberry, let's say, 64554, to the local 554 port. To avoid being asked for credentials, you export the environment variable `DROPBEAR_PASSWORD` and dbclient will use it automagically!
> 
> I do something very similar with the following code inside `debug_cmd.sh`. It has been working like a charm!
> 
> ```
> # Custom Environment Variables.
> export DROPBEAR_PASSWORD=<password>
> 
> # Enable SSH Tunnel.
> while true
> do
>   sleep 60 # Wait 60 secs and repeat every minute.
>   # Check whether dbclient is running.
>   if ! ps | grep "dropbearmulti\sdbclient" >/dev/null
>   then
>     /mnt/mmc01/0/bin/dropbearmulti dbclient -N -f -y -g -R 64554:localhost:554 <user>@<server_ip_or_dns>/<ssh_port>
>   fi
> done &
> ```
> 
> I actually use it to get past a NAT444 that my ISP has deployed on 4G networks. My first option was using a VPN (like WireGuard), but I couldn't find a client already compiled for this architecture (ARMv6), unfortunately.

hi,
I too tinkered with my private IP 4g connection in order to access my local network.. ended up by using raspberry creating a reverse-ssh-tunnel along with another raspberry on a network with public IP.. 

anyway at this time I ended up with a public IP 4g connection so i would avoid using raspberry so for this reason i excluded dropbear trick (already thinked about it..) as raspberry often get stuck so not very reliable for continuative remote access. Another option is to acquire stream and transcode to a different port, also adding security login.

I will try 8001 port thing, seems an easy one.. even if i'd have to set a password (cameras provide passwordless access.. not ideal for remote public access)