ant-thomas/zsgx1hacks #184: GK7205/Hisilicon Hi3516E v200

fss-hacks, 2023-11-24:
Please advise if it is possible to use this with a Goke GK7205. 
People on Russian forums are desperate to know, this chip is now in circulation more than the 7202.
I have a lot of collected information about these cameras, including data on the various apps Closeli (the company running the closed P2P cloud the cameras are forced to connect to) uses, the servers the cameras request etc.
Closeli has caused major security holes for everyone who bought these cameras without knowing they were connected to a closed P2P cloud and I especially want to change this.
I am willing to test any firmwares, translate articles from Russian and Chinese, etc.

fss-hacks, 2024-10-29:
> Hello - do you have dumps or know a place where to find original firmware for gk7202 or gk7205? I have gk7202 camera but it ships with some weird TuyaOS RTOS that doesn't even have u-boot in it, so flashing it requires replacing the whole flash. I'm trying to gather information on how the non-tuya version works.

Someone correct me if I’m wrong but I believe that due to the many, many different lenses and image sensors etc, there isn’t a generic broadly applicable firmware.
You need to identify your device chipset AND what’s attached to it. Try matching the specs of the camera and the mp of the camera and anything you know about it and then trace it back to the original camera manufacturer.
Look at what app it uses to communicate with, the packaging, its MAC identifiers and WiFi name etc.

Determine if the camera has telnet access at all. Then based on the manufscturer start looking for telnet root passwords.

That was my first mistake starting out with this stuff, I thought that generic hacks like this would work.

With a Tuya OS system you should be getting at least a built in RTSP feed if not onvif in my experience.

I wish I could be more help. This is just such a vast universe of devices and firmware is so specific and tailored.

Nicarim, 2024-10-29:
The original TuyaOS was some really specific distribution designed for this particular use case - the camera is powered by an internal battery and the wifi chip on it is responsible for waking up the SoC whenever it recieves traffic. After a while the wifi chip puts the SoC to sleep so there is no way to get any stream out of it.

I was able to dump the flash - there was nothing particulary useful since this RTOS doesnt have file system per se. There were just some files related to the tuya configs and some calibration files for camera but that's about it.

I flashed the uboot from openIPC and then OpenIPC but met with another obstacle - now the wifi chip can't properly communicate with SoC so after a minute or so it turns the SoC off. I will have to either modify the PCB to make it permamently on and use usb power instead - or try to reverse engineer the SDIO protocol it uses to communicate with the wifi chip.

This psrticular camera I have is a peephole style camera - but the original Tuya App it uses is HORRIBLE and it saves the videos to some propetiary format on SD card so I am trying to find a way to make it more sane.

fss-hacks, 2024-10-29:
Ah yeah, that is a nightmare. So in general these firmwares aren’t at all meant for anything battery operated (or at least a device that “thinks” it’s battery operated).

I can see why you went the openipc route.

You are at a hardware level above my technical knowledge. One possibility I have for you is something I’ve done, which is to disconnect the internal battery and replace it with a much larger capacity one that has USB charging but the same power given off along with USB connectivity; like a power bank battery you have removed from the casing but is still attached to the battery pack.

I know you seem willing to go thru the same extreme lengths I am for this stuff but if you are looking for that style camera with more possibilities, the O-Kam series of cameras from Vstarcam has a lot of potential and are very good quality along with a recent security leak.

There is also another small dice sized camera that is a Hopeway device that is Tuya compatible with onvif compatibility that has shown up occasionally on Amazon.

A lot of these devices are built semi clandestinely in China in violation of Chinese law and use unknown cloud servers hosted by unknown actors. Be very, very careful with them for many reasons.

Ph0rk0z, 2024-12-24:
I got a camera that is supported by icam365 software. It says it is GK7202. Not much info on this thing or what it runs. I guess I have to take it apart. These have an external flash chip? Or is flash on SOC? I have a hardware reader so if it's a standard 8 pin chip, I can try to read it out. My camera is A/C powered out of a socket so it shouldn't have any shutdown issues.

Going to have to take it apart and see if there are pins for UART, etc. I don't really want to sign up for an account with their app or any of that crap.

bennettopaul, 2025-04-04:
> Hello - do you have dumps or know a place where to find original firmware for gk7202 or gk7205? I have gk7202 camera but it ships with some weird TuyaOS RTOS that doesn't even have u-boot in it, so [...]

It seems we might be dealing with similar devices.

I'm using a [Cocoon Connect thing](https://cocoonproducts.com.au/shop/home/smarthome/smart-wireless-camera-with-solar-panel-kit/).

It's got a Goke 7202V300 Chipset and it has UART output that mentions RTOS and TUYA. I've noticed the UART pins for the unlabelled WIFI chip (it's got a QR code). The output for the WIFI chip looks similar to a Hi3861. It does seem to put the Goke to sleep.

How did you manage to flash the chip? With the SD card method here? It sounds like the Hi3861 will need it's own firmware.

fss-hacks, 2025-04-12:
> I got a camera that is supported by icam365 software. It says it is GK7202. Not much info on this thing or what it runs. I guess I have to take it apart. These have an external flash chip? Or is flash on SOC? I have a hardware reader so if it's a standard 8 pin chip, I can try to read it out. My camera is A/C powered out of a socket so it shouldn't have any shutdown issues.
> 
> Going to have to take it apart and see if there are pins for UART, etc. I don't really want to sign up for an account with their app or any of that crap.
> 
> edit: I have made a dump of my camera: [PTZ-CAM.zip](https://github.com/user-attachments/files/18241949/PTZ-CAM.zip)
> 
> I can get it online using cls.conf file. But only open ports are 80, 554, 8001 and 8200.
> 
> root password if anyone wants to crack it: root:yE7gW4O0CSXXg:0:0::/root:/bin/sh

I just saw your post. I'm now dealing with an icam 365 that is running on a goke 7205v210. Check out the fork of this hack that is for the Augmetix hc1703, I was actually able to boot busybox and ftp on it. Unfortunately it's a dual lens camera and the RTSP stream is only showing one camera despite there being 2, I thought I might find a way to display both cameras with the hack.